stm32-rs/stm32f1xx-hal #268: Binary size more than 128K

Gustbel, 2020-09-20:
I am experimenting with my BluePill STM32f103, it has 128K memory.
I have a problem and it is that when I generate the binary file that I will load later, I do it in the following way:
`$ cargo build --release --example blink --features stm32f103`
The compilation is done well, but when I go to the "release" folder I find a file close to 260Kbytes.
What am I doing wrong? Why am I generating such large binaries?
Thank you so much

therealprof, 2020-09-20:
The file size size is not indicative of the to-be-flashed size. It most likely contains debuginfo and other stuff.

Use cargo size (after installing cargo-binutils) or arm-none-eabi-size to figure out the real size. BTW: The Bluepill usually features the C8 version of the chip which officially only has 64kB of flash memory.

Gustbel, 2020-09-22:
> The file size size is not indicative of the to-be-flashed size. It most likely contains debuginfo and other stuff.
>
> Use cargo size (after installing cargo-binutils) or arm-none-eabi-size to figure [...]

Thanks for answering me! Is true, with that command I can see the real size that my binary will have. Do you know any method to extract this binary file and save it on my computer without having to need a connected development board? I've been trying a lot and I can't find the command. Thank you very much.

TeXitoi, 2020-09-23:
```
cargo install cargo-binutils
cargo objcopy --bin crate-name --release -- -O binary crate-name.bin
```

Gustbel, 2020-09-23:
> ```
> cargo install cargo-binutils
> cargo objcopy --bin crate-name --release -- -O binary crate-name.bin
> ```

Thanks @TeXitoi !! It's Works! and I was able to download the binary on my STM32 <3
Thanks also to @therealprof

closed 2020-09-23 by Gustbel