stm32-rs/stm32f1xx-hal #269: 16-Bit Spi Frames

BBScholar, 2020-09-21:
_New to Rust_ - I'm struggling to get 16 bit addresses working on the SPI peripheral. I've read the source code and the documentation but cannot get it to work. Any advice? Sorry if I am overlooking something.

Referring to this method within the spi.rs file:
```
/// Converts from 8bit dataframe to 16bit.
    pub fn frame_size_16bit(self) -> Spi<SPI, REMAP, PINS, u16> {
        self.spi.cr1.modify(|_, w| w.spe().clear_bit());
        self.spi.cr1.modify(|_, w| w.dff().set_bit());
        self.spi.cr1.modify(|_, w| w.spe().set_bit());
        Spi {
            spi: self.spi,
            pins: self.pins,
            _remap: PhantomData,
            _framesize: PhantomData,
        }
    }
```

Here's my code:
```
let mut spi =  spi::Spi::spi1(device.SPI1, spi_pins, &mut afio.mapr, spi_mode, 100.khz(), clocks, &mut rcc.apb2);
let spi = spi.frame_size_16bit();
```

Error from `cargo b`:
```
no method named `frame_size_16bit` found for struct `stm32f1xx_hal::spi::Spi<stm32f1xx_hal::pac::SPI1, stm32f1xx_hal::spi::Spi1NoRemap, (stm32f1xx_hal::gpio::gpioa::PA5<stm32f1xx_hal::gpio::Alternate<stm32f1xx_hal::gpio::PushPull>>, stm32f1xx_hal::gpio::gpioa::PA6<stm32f1xx_hal::gpio::Input<stm32f1xx_hal::gpio::Floating>>, stm32f1xx_hal::gpio::gpioa::PA7<stm32f1xx_hal::gpio::Alternate<stm32f1xx_hal::gpio::PushPull>>)>` in the current scope
   --> src/main.rs:191:13
```

thalesfragoso, 2020-09-22:
According to the changelog, this feature isn't release yet on crates.io, if you want to use the git version you can do so by adding something like this to your `Cargo.toml`:

```toml
[patch.crates-io]
stm32f1xx-hal = { git = "https://github.com/stm32-rs/stm32f1xx-hal", features = ["your-mcu", "rt"] }
```

BBScholar, 2020-09-22:
Ah. I didn't think about that. That solution works for me. Thanks for your help!

closed 2020-09-22 by BBScholar