stm32-rs/stm32f1xx-hal #279: SPI no frame format option.

Nekspire, 2020-10-24:
Hi, 
When i was writing driver to some device i've found that HAL implementation does not allow me to set frame format. In my case this is lsbfirst which i have to choose. Can you explain me how can i change it from default msbfirst ?

Nekspire, 2020-10-24:
Okay, i've found solution. 
So what i did: Firstly i downloaded stm32f1xx-hal from repository. Then i added it locally to my project as a workspace and modified spi.rs code:
`        spi.cr1.write(|w| {
            w
                // clock phase from config
                .cpha()
                .bit(mode.phase == Phase::CaptureOnSecondTransition)
                // clock polarity from config
                .cpol()
                .bit(mode.polarity == Polarity::IdleHigh)
                // mstr: master configuration
                .mstr()
                .set_bit()
                // baudrate value
                .br()
                .bits(br)
                // lsbfirst: MSB first
                .lsbfirst()
                .set_bit()
                // ssm: enable software slave management (NSS pin free for other uses)
                .ssm()
                .set_bit()
                // ssi: set nss high = master mode
                .ssi()
                .set_bit()
                // dff: 8 bit frames
                .dff()
                .clear_bit()
                // bidimode: 2-line unidirectional
                .bidimode()
                .clear_bit()
                // both TX and RX are used
                .rxonly()
                .clear_bit()
                // spe: enable the SPI bus
                .spe()
                .set_bit()
        });`
Where i changed the value of **.lsbfirst()** from **.clear_bit()** to **.set_bit()** . Now when spi1 instance is created the lsbfirst bit is set correctly.

burrbull, 2020-10-25:
Yes, this is issue. We need to add more configurable options to init.

TheZoq2, 2020-10-25:
Since this would be a nice extension to have included, I'll re-open the issue :)

PR #280 (2020-11-05): Add SPI frame format config method
Fixes #279

This adds a method to the SPI type which allows configuring which frame format to use for data transfers.
commit a6dde780 (2020-11-05): Add SPI frame format config method

 Fixes #279
closed 2020-11-05 by TheZoq2