stm32-rs/stm32f1xx-hal #286: Examples overflowing flash

fmauNeko, 2020-11-27:
Hello,

I've been trying to build the usb_serial example and run it on a Blue Pill, which is the board being targetted by that example, but I'm getting FLASH overflow errors when building it using the dev profile.
Release fits without issue.
I've been building it with "stm32f103 medium rt stm32-usbd" features.

Is there anything I'm doing blatantly wrong, or is that to be expected with some dependencies having grown?

<details>
<summary>cargo build output</summary>

```
❯ cargo build --example usb_serial --features "stm32f103 medium rt stm32-usbd" --verbose
       Fresh version_check v0.9.2
       Fresh semver-parser v0.7.0
       Fresh stable_deref_trait v1.2.0
       Fresh unicode-xid v0.2.1
       Fresh vcell v0.1.2
       Fresh bitfield v0.13.2
       Fresh autocfg v1.0.1
       Fresh nb v1.0.0
       Fresh void v1.0.2
       Fresh hashbrown v0.9.1
       Fresh r0 v0.2.2
       Fresh usb-device v0.2.7
       Fresh rtfm-core v0.3.0
       Fresh cobs v0.1.4
       Fresh m v0.1.1
       Fresh panic-halt v0.2.0
       Fresh either v1.6.1
       Fresh semver v0.9.0
       Fresh volatile-register v0.2.0
       Fresh embedded-dma v0.1.2
       Fresh nb v0.1.3
       Fresh typenum v1.12.0
       Fresh rustc_version v0.2.3
       Fresh proc-macro2 v1.0.24
       Fresh byteorder v1.3.4
       Fresh embedded-hal v0.2.4
       Fresh crc16 v0.4.0
       Fresh generic-array v0.12.3
       Fresh generic-array v0.14.4
       Fresh generic-array v0.13.2
       Fresh quote v1.0.7
       Fresh generic-array v0.11.1
       Fresh serde v1.0.117
       Fresh indexmap v1.6.0
       Fresh hash32 v0.1.1
       Fresh usbd-serial v0.1.1
       Fresh as-slice v0.1.4
       Fresh syn v1.0.51
       Fresh mfrc522 v0.2.0
       Fresh aligned v0.3.4
       Fresh bare-metal v0.2.5
       Fresh cortex-m-rt-macros v0.1.8
       Fresh rtfm-syntax v0.4.0
       Fresh heapless v0.5.6
       Fresh serde_derive v1.0.117
       Fresh heapless v0.4.4
       Fresh cortex-m v0.6.4
       Fresh cortex-m-rt v0.6.13
       Fresh cortex-m-rtfm-macros v0.5.1
       Fresh cast v0.2.3
       Fresh stm32f1 v0.11.0
       Fresh stm32-usbd v0.5.1
       Fresh cortex-m-semihosting v0.3.5
       Fresh cortex-m-rtfm v0.5.3
       Fresh panic-itm v0.4.2
       Fresh panic-semihosting v0.5.6
warning: unnecessary parentheses around block return value
[... 53 lines cut ...]
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 204 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 228 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 250 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 270 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 284 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 344 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 370 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 458 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 740 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 790 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 802 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 826 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 848 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 872 bytes
          rust-lld: error: too many errors emitted, stopping now (use -error-limit=0 to see all errors)

error: aborting due to previous error; 4 warnings emitted

error: could not compile `stm32f1xx-hal`.

Caused by:
  process didn't exit successfully: `rustc --crate-name usb_serial --edition=2018 examples/usb_serial.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C codegen-units=1 -C debuginfo=2 --cfg 'feature="device-selected"' --cfg 'feature="medium"' --cfg 'feature="rt"' --cfg 'feature="stm32-usbd"' --cfg 'feature="stm32f103"' -C metadata=378dda2e8c9db509 -C extra-filename=-378dda2e8c9db509 --out-dir /home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/examples --target thumbv7m-none-eabi -L dependency=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps -L dependency=/home/fmauneko/stm32f1xx-hal/target/debug/deps --extern byteorder=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libbyteorder-ceb55d0192811e6d.rlib --extern cast=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libcast-dcd38b8550099ad7.rlib --extern cobs=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libcobs-a1e29cc170304221.rlib --extern cortex_m=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libcortex_m-79e5b9b958debf36.rlib --extern cortex_m_rt=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libcortex_m_rt-7695143ad91feebc.rlib --extern rtfm=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/librtfm-cbcf9b49535e1b1f.rlib --extern cortex_m_semihosting=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libcortex_m_semihosting-7a35bb5aedbff9e9.rlib --extern crc16=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libcrc16-b3b569d1ad2f11fa.rlib --extern either=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libeither-9b8bc6ada901b180.rlib --extern embedded_dma=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libembedded_dma-80866b5b86a431bc.rlib --extern embedded_hal=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libembedded_hal-bc44622629a845c4.rlib --extern heapless=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libheapless-6f657b0cd60bc3a5.rlib --extern m=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libm-2b1bf38f2b248799.rlib --extern mfrc522=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libmfrc522-57a2cdf596a1dd23.rlib --extern nb=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libnb-4481428cfdafe9b1.rlib --extern panic_halt=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libpanic_halt-14806ed022f863b8.rlib --extern panic_itm=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libpanic_itm-abcd55b08acf372f.rlib --extern panic_semihosting=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libpanic_semihosting-6204078b9591b19b.rlib --extern serde=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libserde-6dc9250bd9ac23c6.rlib --extern serde_derive=/home/fmauneko/stm32f1xx-hal/target/debug/deps/libserde_derive-4969e75874363f4b.so --extern stm32_usbd=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libstm32_usbd-6c5c8737f91ffef5.rlib --extern stm32f1=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libstm32f1-80fdec1cedbde09f.rlib --extern stm32f1xx_hal=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libstm32f1xx_hal-bcb921448bf1f309.rlib --extern usb_device=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libusb_device-396a45694bec867f.rlib --extern usbd_serial=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libusbd_serial-0d945700cb982e90.rlib --extern void=/home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/deps/libvoid-25838a15faf9c129.rlib -C link-arg=-Tlink.x -L /home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/build/cortex-m-52cb6c434e0a1e7d/out -L /home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/build/cortex-m-rt-0639961e2147c568/out -L /home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/build/cortex-m-semihosting-813c4f3414df18fe/out -L /home/fmauneko/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/build/stm32f1-794041ac46229d23/out` (exit code: 1)
```

</details>

TeXitoi, 2020-11-27:
Debug mode are, most of the time, too big, so this is expected.

TheZoq2, 2020-12-01:
Yea, for large projects you'll almost always want to run in release mode, we should probably add a note about that to the example

zacck, 2021-03-26:
@TheZoq2  would be acceptable to add a note as comments above the code?

TheZoq2, 2021-04-06:
> @TheZoq2 would be acceptable to add a note as comments above the code?

Yes, that seems like a great idea. Would you like to submit a PR for that, if not, I can do it at some point

zacck, 2021-04-06:
@TheZoq2  will throw in a PR after work today or tomorrow, just wanted to know if it would be relevant work.

PR #326 (2021-04-08): USB-SERIAL_NOTES: add note to inform user to build in release mode
Fixes #286 

## Changes Proposed 
- [x] Add note on why one needs to build the example using the release profile
closed 2021-04-08 by TheZoq2