stm32-rs/stm32f1xx-hal #287: no method named `spi3_remap` found

ohmyarch, 2020-11-29:
error message:
```
   Compiling stm32f1xx-hal v0.7.0
error[E0599]: no method named `spi3_remap` found for mutable reference `&mut stm32f1::W<u32, Reg<u32, _MAPR>>` in the current scope
   --> C:\Users\xxx\scoop\persist\rustup\.cargo\registry\src\github.com-1ecc6299db9ec823\stm32f1xx-hal-0.7.0\src\spi.rs:226:35
    |
226 |         mapr.modify_mapr(|_, w| w.spi3_remap().bit(REMAP::REMAP));
    |                                   ^^^^^^^^^^ help: there is an associated function with a similar name: `spi1_remap`
```

Cargo.toml:
```toml
stm32f1xx-hal = { version = "0.7.0", features = ["rt", "stm32f103", "high"] }
```

target MCU:
STM32F103ZET6

burrbull, 2020-11-29:
`spi3_remap` is defined for `connectivity` devices only:

https://github.com/stm32-rs/stm32f1xx-hal/blob/b3e2d75b1e9d7e4398304855d4e2ae892c745fc7/src/spi.rs#L212-L213

Enable `connectivity` feature if your device support this.

ohmyarch, 2020-11-30:
"connectivity" is for stm32f105xx and stm32f107xx, as I mentioned, I'm using a stm32f103zet6, only "high" is needed. `features = ["rt", "stm32f103", "high"]` work fine with stm32f1xx-hal 0.6.1 but not with 0.7.0.

ohmyarch, 2020-12-03:
@burrbull What do you think?

ohmyarch, 2021-01-18:
Any follow-up?

YuaXan, 2021-03-04:
I used new commit<0f367e6>, and could not compile stm32f1xx-hal too.
```
Checking stm32f1xx-hal v0.7.0 (https://github.com/stm32-rs/stm32f1xx-hal#0f367e60)
error[E0599]: no method named `spi3_remap` found for mutable reference `&mut stm32f1::W<u32, Reg<u32, _MAPR>>` in the current scope
   --> C:\Users\92307\.cargo\git\checkouts\stm32f1xx-hal-bead1e0eb6c95bf1\0f367e6\src\spi.rs:239:35
    |
239 |         mapr.modify_mapr(|_, w| w.spi3_remap().bit(REMAP::REMAP));
    |                                   ^^^^^^^^^^ help: there is an associated function with a similar name: `spi1_remap`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0599`.
error: could not compile `stm32f1xx-hal`

To learn more, run the command again with --verbose.
```

YuaXan, 2021-03-04:
I used new commit<0f367e6>, and it could not compile `stm32f1xx-hal`.
```
Checking stm32f1xx-hal v0.7.0 (https://github.com/stm32-rs/stm32f1xx-hal#0f367e60)
error[E0599]: no method named `spi3_remap` found for mutable reference `&mut stm32f1::W<u32, Reg<u32, _MAPR>>` in the current scope
   --> C:\Users\xxxxx\.cargo\git\checkouts\stm32f1xx-hal-bead1e0eb6c95bf1\0f367e6\src\spi.rs:239:35
    |
239 |         mapr.modify_mapr(|_, w| w.spi3_remap().bit(REMAP::REMAP));
    |                                   ^^^^^^^^^^ help: there is an associated function with a similar name: `spi1_remap`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0599`.
error: could not compile `stm32f1xx-hal`
```

PR #312 (2021-03-14): Fix SPI3 alternate function remapping.
The remap is available only in connectivity line devices.
Closes #287
https://www.st.com/resource/en/reference_manual/cd00171190-stm32f101xx-stm32f102xx-stm32f103xx-stm32f105xx-and-stm32f107xx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf
[image]
closed 2021-03-14 by TheZoq2