stm32-rs/stm32f1xx-hal #288: impl GpioExt for GPIOF/GPIOG?

ohmyarch, 2020-12-03:
Seems to be missing.

ohmyarch, 2020-12-03:
Just found that there are no modules for gpiof/gpiog.

TheZoq2, 2021-01-23:
Good catch (and sorry for the delay), I don't think there is a good reason for us not having that already. If you're interested, I could show you how to add the impls

xieren58, 2021-02-01:
How to add the impls ? (for gpiof/gpiog...)

xieren58, 2021-02-01:
@TheZoq2

TheZoq2, 2021-02-08:
Sorry for the delay. I think all that is required is to add another one of these macros https://github.com/stm32-rs/stm32f1xx-hal/blob/master/src/gpio.rs#L1315 for the other peripherals. It probably also needs feature gating because IIRC those ports are only available on some of the chips

xieren58, 2021-02-21:
@TheZoq2 Thanks!

codytrey, 2021-04-16:
Seems that additional timers need to be implemented to fully use GPIOF/G. Since that only affects things like pwm, and adc, should this be it's own issue?

ohmyarch, 2021-06-23:
Any progress?