stm32-rs/stm32f1xx-hal #301: RX DMA with USART seems to go into an infinite loop

zacck, 2020-12-31:
I am building an application for a STM32F103CT6 board and I need to use serial communication. However, whenever I attempt to receive data on USART with DMA the operation seems to go into an infinite loop. 

When I use `.wait()` and observe the application via gdb the registers show that I am stuck in the wait call see below 

``` code 
   // Set up the usart device. Taks ownership over the USART register and tx/rx pins. The rest of
    // the registers are used to enable and configure the device.
    let serial = Serial::usart3(
        p.USART3,
        (tx, rx),
        &mut afio.mapr,
        Config::default().baudrate(9_600.bps()),
        clocks,
        &mut rcc.apb1,
    );

    //DMA UART
    let (tx, rx) = serial.split();
    let tx_dma = tx.with_dma(channels.2);
    let sent = b"Z";
    tx_dma.write(sent).wait();
    let rx_dma = rx.with_dma(channels.3);
    let buff = singleton!(: [u8; 8] = [0; 8]).unwrap();
    let (first, _) = rx_dma.read(buff).wait();
    dbg!(first);
 ```
 
 ```
 (gdb) backtrace
#0  stm32f1::generic::Reg<U,REG>::read (self=0x40020000)
    at /Users/zacck/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f1-0.11.0/src/generic.rs:52
#1  stm32f1xx_hal::dma::dma1::C3::isr (self=0x20004d4c)
    at /Users/zacck/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f1xx-hal-0.6.1/src/dma.rs:208
#2  0x08004c1c in stm32f1xx_hal::dma::dma1::C3::in_progress (self=0x20004d4c)
    at /Users/zacck/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f1xx-hal-0.6.1/src/dma.rs:177
#3  0x08000fd8 in stm32f1xx_hal::dma::dma1::<impl stm32f1xx_hal::dma::Transfer<MODE,BUFFER,stm32f1xx_hal::dma::RxDma<PAYLOAD,stm32f1xx_hal::dma::dma1::C3>>>::is_done (self=0x20004d4c)
    at /Users/zacck/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f1xx-hal-0.6.1/src/dma.rs:302
#4  0x08000f7c in stm32f1xx_hal::dma::dma1::<impl stm32f1xx_hal::dma::Transfer<MODE,BUFFER,stm32f1xx_hal::dma::RxDma<PAYLOAD,stm32f1xx_hal::dma::dma1::C3>>>::wait (self=...)
    at /Users/zacck/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f1xx-hal-0.6.1/src/dma.rs:306
#5  0x080015a0 in chungabox::__cortex_m_rt_main () at src/main.rs:87
#6  0x080012e4 in main () at src/main.rs:24
(gdb) info registers
r0             0x0                 0
r1             0x0                 0
r2             0x1080              4224
r3             0x20004cd4          536890580
r4             0x7a1200            8000000
r5             0x7a1200            8000000
r6             0x7a1200            8000000
r7             0x20004d08          536890632
r8             0x4                 4
r9             0x80083ac           134251436
r10            0x0                 0
r11            0x0                 0
r12            0xf4240             1000000
sp             0x20004cf0          0x20004cf0
lr             0x8004c87           134237319
pc             0x8004c88           0x8004c88 <stm32f1xx_hal::dma::dma1::C3::isr+30>
xPSR           0x61000000          1627389952
msp            0x20004cf0          0x20004cf0
psp            0x0                 0x0
primask        0x0                 0
basepri        0x0                 0
faultmask      0x0                 0
control        0x0                 0
```

When I use the [peek](https://github.com/stm32-rs/stm32f1xx-hal/blob/master/examples/serial-dma-peek.rs) function after the call to read I notice the slice is always the first byte. 

I should mention that receiving and sending over USART without DMA works, Sending with DMA works and only receiving with DMA is broken 

Could anyone help me figure this out or point me in the right direction to work on a fix, thank you!

burrbull, 2020-12-31:
The first that I can propose is to check this repository on regression.

zacck, 2020-12-31:
@burrbull  to check on regression of the feature? and how would I do that? is there a test suite I can use?

burrbull, 2020-12-31:
I mean try old versions of this library.

zacck, 2020-12-31:
@burrbull  ok let me try the last few releases

zacck, 2020-12-31:
@burrbull  I just tried version 0.4.0 which seem to be the oldest I can try yet, the  application still hangs even although the error is a bit different, here  is my backtrace and registers 

```
(gdb) backtrace
#0  0x0800420c in stm32f1xx_hal::dma::dma1::C3::isr (self=0x20000001 <chungabox::__cortex_m_rt_main::{{closure}}::VAR+1>)
    at /Users/zacck/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f1xx-hal-0.4.0/src/dma.rs:206
#1  0x0800418c in stm32f1xx_hal::dma::dma1::C3::in_progress (self=0x20004d44)
    at /Users/zacck/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f1xx-hal-0.4.0/src/dma.rs:177
#2  0x08001ce0 in stm32f1xx_hal::dma::dma1::<impl stm32f1xx_hal::dma::Transfer<MODE,BUFFER,stm32f1xx_hal::dma::RxDma<PAYLOAD,stm32f1xx_hal::dma::dma1::C3>>>::is_done (self=0x20004d44)
    at /Users/zacck/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f1xx-hal-0.4.0/src/dma.rs:302
#3  0x08001c84 in stm32f1xx_hal::dma::dma1::<impl stm32f1xx_hal::dma::Transfer<MODE,BUFFER,stm32f1xx_hal::dma::RxDma<PAYLOAD,stm32f1xx_hal::dma::dma1::C3>>>::wait (self=...)
    at /Users/zacck/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f1xx-hal-0.4.0/src/dma.rs:306
#4  0x08002398 in chungabox::__cortex_m_rt_main () at src/main.rs:87
#5  0x080020d6 in main () at src/main.rs:24
(gdb) info registers
r0             0x20004d44          536890692
r1             0x0                 0
r2             0x1080              4224
r3             0x0                 0
r4             0x7a1200            8000000
r5             0x7a1200            8000000
r6             0x7a1200            8000000
r7             0x20004d28          536890664
r8             0x4                 4
r9             0x80086a4           134252196
r10            0x0                 0
r11            0x0                 0
r12            0x20004f04          536891140
sp             0x20004cf8          0x20004cf8
lr             0x800418d           134234509
pc             0x800420c           0x800420c <stm32f1xx_hal::dma::dma1::C3::isr+2>
xPSR           0x61000000          1627389952
msp            0x20004cf8          0x20004cf8
psp            0x0                 0x0
primask        0x0                 0
basepri        0x0                 0
faultmask      0x0                 0
control        0x0                 0
(gdb)
```

burrbull, 2020-12-31:
One question about your example.
What data you send to RX?
Looks like USART DMA just doesn't get any data and stays in `wait`.

zacck, 2020-12-31:
@burrbull  I am trying to send a string as bytes here is my code at the moment 

```
 // the registers are used to enable and configure the device.
    let serial = Serial::usart3(
        p.USART3,
        (tx, rx),
        &mut afio.mapr,
        9600.bps(),
        clocks,
        &mut rcc.apb1,
    );

    //DMA UART
    let (tx, rx) = serial.split();
    let tx_dma = tx.with_dma(channels.2);
    let sent = b"Z";
    tx_dma.write(sent).wait();
    let rx_dma = rx.with_dma(channels.3);
    let buff = singleton!(: [u8; 8] = [0; 8]).unwrap();
    let (first, _) = rx_dma.read(buff).wait();
    dbg!(first);
    ````

zacck, 2020-12-31:
I have the two pins connected, so I  can test

burrbull, 2020-12-31:
If you try to send `b"Z"` from TX to RX of same board, you can see that you've sent data before you even create and start receiver.

zacck, 2020-12-31:
@burrbull  so should I create and start the receiver before sending?

burrbull, 2020-12-31:
Yes, `wait` is blocking operation. It will block sending.
So you can do loopback with DMA only by splitting TX and RX to different tasks (for example with RTIC).
Or by checking DMA state by hand.

zacck, 2020-12-31:
@burrbull  here is how I have modified the code. 

```
  // Set up the usart device. Taks ownership over the USART register and tx/rx pins. The rest of
    // the registers are used to enable and configure the device.
    let serial = Serial::usart3(
        p.USART3,
        (tx, rx),
        &mut afio.mapr,
        9600.bps(),
        clocks,
        &mut rcc.apb1,
    );

    //DMA UART
    let (tx, rx) = serial.split();
    let rx_dma = rx.with_dma(channels.3);
    let buff = singleton!(: [u8; 8] = [0; 8]).unwrap();
    let tx_dma = tx.with_dma(channels.2);
    let sent = b"Z";
    let read = rx_dma.read(buff);
    tx_dma.write(sent).wait();
    //let (first, _) = rx_dma.read(buff).wait();
    while !read.is_done() {
        let slice = read.peek();
        dbg!(slice);
    }
    //dbg!(first);
    loop {}
 ```

I have been splitting the TX ans RX  its the line after  I configure my serial, or is there a further split I need to do?

This works however I get stuck in the while loop, is there some rust I dont know that I should be doing below is my openocd and gdb 

```openocd 
 ✗ openocd
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v29 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.170642
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection on tcp/3333
Info : device id = 0x20036410
Info : flash size = 128kbytes
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08000130 msp: 0x20005000
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000003a msp: 0x20005000
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08000130 msp: 0x20005000
semihosting is enabled
Info : halted: PC: 0x08000bde
[src/main.rs:58] "Boot successful" = "Boot successful"
[src/main.rs:91] slice = []
[src/main.rs:91] slice = [
    90,
]
[src/main.rs:91] slice = [
[... 72 lines cut ...]
(gdb) info registers
r0             0x5                 5
r1             0x20004bac          536890284
r2             0x3                 3
r3             0x5                 5
r4             0x0                 0
r5             0x1                 1
r6             0x8008b4c           134253388
r7             0x20004bd8          536890328
r8             0x20004f58          536891224
r9             0x10                16
r10            0x50                80
r11            0x4                 4
r12            0x20004bac          536890284
sp             0x20004b60          0x20004b60
lr             0x8002959           134228313
pc             0x800317c           0x800317c <__c_m_sh_syscall>
xPSR           0x21000000          553648128
msp            0x20004b60          0x20004b60
psp            0x0                 0x0
primask        0x1                 1
basepri        0x0                 0
faultmask      0x0                 0
control        0x0                 0
```

burrbull, 2020-12-31:
Does this work?
```rust
    let read = rx_dma.read(buff);
    tx_dma.write(sent).wait();
    read.wait();
    dbg!(first);
    loop {}
```

zacck, 2020-12-31:
@burrbull  this still gets stuck at `read.wait()` 

Here is my backtrace and registers 

```
(gdb) backtrace
#0  stm32f1::stm32f103::dma1::isr::TCIF1R::bit_is_clear (self=0x8bac)
    at /Users/zacck/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f1-0.7.1/src/stm32f103/dma1/isr/mod.rs:72
#1  0x08003ed2 in stm32f1xx_hal::dma::dma1::C3::in_progress (self=0x20004d94)
    at /Users/zacck/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f1xx-hal-0.3.0/src/dma.rs:172
#2  0x080015f2 in stm32f1xx_hal::dma::dma1::<impl stm32f1xx_hal::dma::Transfer<MODE,BUFFER,stm32f1xx_hal::dma::RxDma<PAYLOAD,stm32f1xx_hal::dma::dma1::C3>>>::is_done (self=0x20004d94)
    at /Users/zacck/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f1xx-hal-0.3.0/src/dma.rs:284
#3  0x08001596 in stm32f1xx_hal::dma::dma1::<impl stm32f1xx_hal::dma::Transfer<MODE,BUFFER,stm32f1xx_hal::dma::RxDma<PAYLOAD,stm32f1xx_hal::dma::dma1::C3>>>::wait (self=...)
    at /Users/zacck/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f1xx-hal-0.3.0/src/dma.rs:288
#4  0x08000e22 in chungabox::__cortex_m_rt_main () at src/main.rs:93
#5  0x08000bdc in main () at src/main.rs:24
(gdb) info registers
r0             0x20004d6f          536890735
r1             0x0                 0
r2             0x0                 0
r3             0x1090              4240
r4             0x7a1200            8000000
r5             0x7a1200            8000000
r6             0x7a1200            8000000
r7             0x20004d78          536890744
r8             0x4                 4
r9             0x8008b0c           134253324
r10            0x0                 0
r11            0x0                 0
r12            0xf4240             1000000
sp             0x20004d68          0x20004d68
lr             0x8003ed3           134233811
pc             0x8004252           0x8004252 <stm32f1::stm32f103::dma1::isr::TCIF1R::bit_is_clear>
xPSR           0x61000000          1627389952
msp            0x20004d68          0x20004d68
psp            0x0                 0x0
primask        0x0                 0
basepri        0x0                 0
faultmask      0x0                 0
control        0x0                 0
(gdb)
```

zacck, 2020-12-31:
Noob error, 

```
DMA is usually used to receive an exact number of transfers, that API will consider the buffer size as the number of transfers

The usart only has a 1-byte fifo, so if you receive 2 bytes before reading the first you will get an overrun error and the usart will stop sending dma requests
```
To quote @thalesfragoso  from the stm32-rs  matrix chat

closed 2020-12-31 by zacck