stm32-rs/stm32f1xx-hal #304: UB in SPI and Serial code

TheZoq2, 2021-01-14:
@repnop pointed out to me that https://github.com/stm32-rs/stm32f1xx-hal/blob/master/src/spi.rs#L261 and the `write_volatile`call in `serial.rs`  is UB because constructing (or possibly writing) from a pointer created from a non-mut reference is UB.

A fix might be to just change it to 
```rust
unsafe { ptr::write_volatile(&mut self.spi.dr as *const _ as *mut FrameSize, data) }
```

But spi.dr does not implement `DerefMut` so this fix does not compile. I don't have time to look into it deeper than that today but we should be aware of this potential UB.

It is possible that the whole svd2rust system, suffers from a similar issue, but I'm read up enough on this to know the details. There is apparently some more context in https://github.com/rust-lang/unsafe-code-guidelines/issues/265

chorman0773, 2021-01-15:
I'd be careful with references and volatile (I came here from the issue you, presumably accidentally, linked). As brought up in https://github.com/rust-lang/unsafe-code-guidelines/issues/265, the compiler can invent reads to them at any time, for any or no reason. The raw_ref_macros may solve that issue though.

thalesfragoso, 2021-01-16:
All registers are wrappers around `VolatileCell`, which is a wrapper around `UnsafeCell`, doesn't that prevent the UB ?

repnop, 2021-01-16:
@thalesfragoso for more information about that, see the linked thread in the comment above yours. The tl;dr is more or less: that for MMIO, it is unsound since the compiler is allowed to add spurious reads (as long as they don't cause data races) for `&UnsafeCell` since in the LLVM IR references are always marked as `dereferencable`, which could have side effects unlike normal memory reads. This _currently_ works because `rustc` isn't inserting these, but may in the future. There's more discussion on how to solve this problem in the linked issue, but this specific bug is unrelated to that and when Zoq made it I didn't adequately make that clear since some others and I were discussing the aforementioned problem as well as this one at the same time :sweat_smile:

thalesfragoso, 2021-01-16:
I don't get it, the spurious read from MMIO is indeed a problem (although I don't think is UB) that exists everywhere in the ecosystem, not just in the code linked in this issue. However, I think the issue in discussion here isn't that, but the writing to a memory location derived from an immutable reference, no ?

repnop, 2021-01-16:
yes, which is what I said in the latter part of my comment

thalesfragoso, 2021-01-16:
Oh, ok. My comment was addressing this issue btw, not the MMIO one.

repnop, 2021-01-17:
Oh, sorry, I misunderstood then. Going from `&T` to `*const T` to `*mut T` and then writing through it is most definitely undefined behavior, some references:

https://github.com/rust-lang/rust-clippy/issues/4774
https://github.com/rust-lang/unsafe-code-guidelines/issues/257
https://github.com/rust-lang/rfcs/pull/2582

thalesfragoso, 2021-01-18:
> Going from &T to *const T to *mut T and then writing through it is most definitely undefined behavior

With the exception of `UnsafeCell`. In fact, the `UnsafeCell::get()` method is exactly that. That's why I mentioned it up there.

repnop, 2021-01-18:
Ah, I gotcha now, it wasn't immediately obvious there was `UnsafeCell` involved since all of that code is autogenerated and hard to find the actual struct definitions. Guess I was mistaken then, sorry about that! :sweat_smile:

closed 2021-01-19 by TheZoq2