stm32-rs/stm32f1xx-hal #309: How to write generic SPI DMA code?

pozix604, 2021-02-11:
I do not understand how to write generic SPI DMA code with the HAL as defined. I present a minimal example below.

Assume the following `use` declarations:

```
  use stm32f1xx_hal::dma::dma1::*;
  use stm32f1xx_hal::dma::*;
  use stm32f1xx_hal::gpio::gpioa::*;
  use stm32f1xx_hal::gpio::*;
  use stm32f1xx_hal::prelude::*;
  use stm32f1xx_hal::spi::*;
```

The following dummy device driver compiles nicely:

```rust
struct DeviceDriver<SPI> {
    spi: Option<SPI>,
    buf: Option<&'static mut [u8]>,
}

impl<SPI, REMAP, PINS> DeviceDriver<SpiTxDma<SPI, REMAP, PINS, C5>>
where
    SpiTxDma<SPI, REMAP, PINS, C5>: WriteDma<&'static mut [u8], u8>,
{
    pub fn new(spi: SpiTxDma<SPI, REMAP, PINS, C5>, buf: &'static mut [u8]) -> Self {
        Self {
            spi: Some(spi),
            buf: Some(buf),
        }
    }

    pub fn send(&mut self) {
        if let (Some(spi), Some(buf)) = (self.spi.take(), self.buf.take()) {
            let txdma = spi.write(buf);
            let (buf, spi) = txdma.wait();
            self.buf = Some(buf);
            self.spi = Some(spi);
        }
    }
}
```

However, the following does not compile, failing on the `wait`:

```rust
impl<SPI, REMAP, PINS, CH> DeviceDriver<SpiTxDma<SPI, REMAP, PINS, CH>>
where
    SpiTxDma<SPI, REMAP, PINS, CH>: WriteDma<&'static mut [u8], u8>,
{
    pub fn new(spi: SpiTxDma<SPI, REMAP, PINS, CH>, buf: &'static mut [u8]) -> Self {
        Self {
            spi: Some(spi),
            buf: Some(buf),
        }
    }

    pub fn send(&mut self) {
        if let (Some(spi), Some(buf)) = (self.spi.take(), self.buf.take()) {
            let txdma = spi.write(buf);
            let (buf, spi) = txdma.wait();
            self.buf = Some(buf);
            self.spi = Some(spi);
        }
    }
}
```

The error message is:

```
44 |             let (buf, spi) = txdma.wait();
   |                                    ^^^^ method not found in `stm32f1xx_hal::dma::Transfer<stm32f1xx_hal::dma::R, &mut [u8], stm32f1xx_hal::dma::TxDma<stm32f1xx_hal::spi::SpiPayload<SPI, REMAP, PINS>, CH>>`
```

The difference is `CH`. That is, in the first example, I've hardcoded `C5` as the channel to use. In the second example, it is a type parameter called `CH`. There is no other difference.

I just don't get it. What am I missing?

pozix604, 2021-02-12:
As an aside, simplifying the code to the following doesn't work either:

```rust
struct DeviceDriver<DMA> {
    dma: Option<DMA>,
    buf: Option<&'static [u8]>,
}

impl<DMA> DeviceDriver<DMA>
where
    DMA: WriteDma<&'static [u8], u8>,
{
    pub fn send(&mut self) {
        if let (Some(dma), Some(buf)) = (self.dma.take(), self.buf.take()) {
            let transfer = dma.write(buf);
            let (buf, dma) = transfer.wait();
            self.buf = Some(buf);
            self.dma = Some(dma);
        }
    }
}
```

even though the only trait bound is `B: StaticReadBuffer<Word = TS>` which is clearly satisfied by `&'static [u8]`.

Cat-Ion, 2021-06-07:
Timers also don't implement a common trait, so exchanging them is a bit fiddly (at least to me as a complete rust newbie). Other peripherals are probably similar?

Windfisch, 2021-08-17:
@pozix604 I ran into the same problem and would love to see this fixed. Could you show in code how this could be solved?