stm32-rs/stm32f1xx-hal #310: USB access on stm32f105 series

BBScholar, 2021-02-16:
Is there a reason the usb module is disabled for stm32f105 series MCUs? The datasheet says it supports USB OTG FS.
```
#[cfg(all(
    feature = "stm32-usbd",
    any(feature = "stm32f102", feature = "stm32f103")
))]
pub mod usb;
```
I'm relatively new to embedded, so forgive me if this is a dumb question.

burrbull, 2021-02-18:
I think the main reason is connectivity devices are not tested enough.
Also [stm32-usbd](https://github.com/stm32-rs/stm32-usbd) uses its own register map (`pac`) so the relevant place for this issue is there.

TeXitoi, 2021-02-18:
I think that the silicon for the usb part of 105 is not the same as the silicon for the others, as 105 support OTG, but not the others. Thus, that's a different driver that is needed. The needed driver might be https://crates.io/crates/synopsys-usb-otg

@Disasm what do you think?

Disasm, 2021-02-18:
That's true, `synopsys-usb-otg` should work with this chip, but I haven't tried this pair yet.

WagnerPMC, 2024-03-03:
> That's true, `synopsys-usb-otg` should work with this chip, but I haven't tried this pair yet.

Hello,
Have you tried to pair synopsis-usb-otg with F105/F107?