stm32-rs/stm32f1xx-hal #314: Understanding code which uses nb

katyo, 2021-03-04:
I have a problem with understanding code which uses `nb` crate. In particular i2c's `write_read` looks like incorrect. If I understand right this acts like classic Knuth's coroutine. So it simply shuld be called again when `WouldBlock` returned. But Knuth's coroutines has well known problems with state tracking on suspend/resume, and I absolutely does not see any state tracking in current code. Say when I call i2c's `write_read` again after `WouldBlock` is returned it actually starts from the beginning but not resumes.
This code looks like having dose of black magic or completely misfunctional.
Can everyone explain me how it actually works?

TheZoq2, 2021-03-14:
You may be right in that it re-starts the transmission process if called multiple times, at least in some cases. I'm not familiar with Knuth's coroutines, but I see the `nb` stuff more as "the device is busy-" or "nothing is available right now-" "try again later" rather than "this process is not done yet, continue later".

katyo, 2021-03-14:
Current implementation of blocking i2c does not works for me at all. My code interacts with RTC DS3231: It just reads several registers via i2c. But self-written implementation, which operates with registers directly, works fine. I think we should introduce some state management to hold checkpoint and correctly resume I/O after returning `WouldBlock`.