stm32-rs/stm32f1xx-hal #317: Maybe should add Alternate Function Input Mode

YuaXan, 2021-03-07:
I found `Alternate Function Input Mode` from `stm32f1xx_hal_gpio.h`
```c
#define  GPIO_MODE_INPUT                        0x00000000u   /*!< Input Floating Mode                   */
/* codes */
#define  GPIO_MODE_AF_INPUT                     GPIO_MODE_INPUT          /*!< Alternate Function Input Mode         */
```
Although `Alternate Function Input Mode` same `Input Floating Mode `, but I think should add this mode that would be helping user to aviod ambiguity and as same as possible with ` stm32f1xx_hal` Clang vision.

TheZoq2, 2021-03-14:
Does this solve a use case that isn't solved by the combination of `Input` and `AlternateFunction` pin modes already present?

YuaXan, 2021-03-14:
@TheZoq2 In fact, used `Input Floating Mode` same to used `Alternate Input`

Windfisch, 2021-08-17:
I don't like this idea, because from my understanding, there simply is no difference. I would prefer to keep the type-stated I/O pins as close as possible to the actual hardware state transitions.

[The reference manual]() states in 9.1:
> Subject to the specific hardware characteristics of each I/O port listed in the datasheet, each port bit of the General Purpose IO (GPIO) Ports, can be individually configured by software in several modes:
> - Input floating
> - Input pull-up
> - Input-pull-down
> - Analog
> - Output open-drain
> - Output push-pull
> - Alternate function push-pull
> - Alternate function open-drain

It does not mention a "Alternate input" mode, and figure 13 suggests that the after-schmitt-trigger-input-signal is just both routed to the input of an alternate function peripheral *and* to the input register file at the same time.

TheZoq2, 2021-08-18:
That's a very good point, I'll close this issue for now, feel free to re-open if there are other considerations

closed 2021-08-18 by TheZoq2