stm32-rs/stm32f1xx-hal #321: QEI: Unable to (safely) define encoder pins as pull-up

FLamparski, 2021-03-14:
I have this bit of code, which attempts to enable pull-up resistors on PB6 and PB7, and then construct a quadrature encoder interface on top of those pins using TIM4:

```rust
        let enc_a = gpiob.pb6.into_pull_up_input(&mut gpiob.crl);
        let enc_b = gpiob.pb7.into_pull_up_input(&mut gpiob.crl);
        let encoder = Timer::tim4(dp.TIM4, &clocks, &mut rcc.apb1).qei(
            (enc_a, enc_b),
            &mut afio.mapr,
            QeiOptions::default(),
        );
```

This fails because apparently the `qei()` method expects the pins to be floating instead:

```
error[E0277]: the trait bound `PB6<Input<PullUp>>: stm32f1xx_hal::gpio::Mode<Input<Floating>>` is not satisfied
  --> src\main.rs:95:68
   |
95 |         let encoder = Timer::tim4(dp.TIM4, &clocks, &mut rcc.apb1).qei(
   |                                                                    ^^^ the trait `stm32f1xx_hal::gpio::Mode<Input<Floating>>` is not implemented for `PB6<Input<PullUp>>`
   |
   = help: the following implementations were found:
             <PB6<MODE> as stm32f1xx_hal::gpio::Mode<MODE>>
   = note: required because of the requirements on the impl of `stm32f1xx_hal::pwm_input::Pins<Tim4NoRemap>` for `(PB6<Input<PullUp>>, PB7<Input<PullUp>>)`

error[E0277]: the trait bound `PB7<Input<PullUp>>: stm32f1xx_hal::gpio::Mode<Input<Floating>>` is not satisfied
  --> src\main.rs:95:68
   |
95 |         let encoder = Timer::tim4(dp.TIM4, &clocks, &mut rcc.apb1).qei(
   |                                                                    ^^^ the trait `stm32f1xx_hal::gpio::Mode<Input<Floating>>` is not implemented for `PB7<Input<PullUp>>`
   |
   = help: the following implementations were found:
             <PB7<MODE> as stm32f1xx_hal::gpio::Mode<MODE>>
   = note: required because of the requirements on the impl of `stm32f1xx_hal::pwm_input::Pins<Tim4NoRemap>` for `(PB6<Input<PullUp>>, PB7<Input<PullUp>>)`
```

Is there a way to do this safely and within the confines of the library? Note that this configuration seems to be valid as I have tested it using a C program generated in STM32CubeIDE and the behaviour is as I expect it.

FLamparski, 2021-03-14:
I'm guessing the unsafe workaround looks like this:

```rust
        // let enc_a = gpiob.pb6.into_pull_up_input(&mut gpiob.crl);
        unsafe { PB6::<Input<PullUp>>::set_mode(&mut gpiob.crl) };
        // let enc_b = gpiob.pb7.into_pull_up_input(&mut gpiob.crl);
        unsafe { PB7::<Input<PullUp>>::set_mode(&mut gpiob.crl) };
```

Windfisch, 2021-08-05:
Seems like a bug in `pwm_input.rs`, which implements `Pins<REMAP>` only for those GPIOs whose `Mode = Input<Floating>`. I guess it could work to replace the `Floating` by two `PullState1/2` type parameters, so it accepts just any `Input<Whatever>`.

Do you see any problems with that?

burrbull, 2021-08-05:
Sorry. I've not seen this issue. What all allowed pin modes? Are they same for `PwmPin`?

Windfisch, 2021-08-05:
I am gonna look into it and will open a PR this weekend if I find the time

Windfisch, 2021-08-08:
From reading the reference manual, I cannot see a reason why any pullup/down/floating restrictions should be made on any "alternate function" input pin (which aren't really in any "alternate function" mode on this chip).
Also, I could not see any reason why any peripheral that wants to drive an Alternate pin would make any restrictions on the pin being Push-Pull or OpenDrain.

For this cause, I would assume that `Input<Mode>` is valid for any Mode. PR incoming.

Windfisch, 2021-08-08:
I've created the PRs #357, #359 and #358, which incrementally lift the `Input<...>` and `Alternate<...>` restrictions for most peripherals. (I did not touch USB, and I2C remains OpenDrain-only).

Please prefer to review and merge #358.

commit afd80871 (2021-08-17): Merge #358

358: Allow pullup inputs and opendrain outputs r=burrbull a=Windfisch

This fixes #321, #199 and extends on #118, and also fixes this issue for other peripherals that don't mind which pull-ups are configured for their inputs and whether their outputs are pushpull or open drain.

This is a superset of #357 and #359. Prefer this PR.
PR #357 (2021-08-17): Allow PullUp and PullDown inputs for timer pwm_input and qei
This fixes #321.

This PR is a subset of #358 and #359.
PR #358 (2021-08-17): Allow pullup inputs and opendrain outputs
This fixes #321, #199 and extends on #118, and also fixes this issue for other peripherals that don't mind which pull-ups are configured for their inputs and whether their outputs are pushpull or open drain.

This is a superset of #357 and #359. Prefer this PR.
PR #359 (2021-08-17): Allow all pullup inputs
This fixes #321, extends on #118, and also fixes this issue for other peripherals that don't mind whether their Inputs have pull-ups or -downs enabled.

This is a superset of #357 and a subset of #358.
closed 2021-08-17 by bors[bot]