stm32-rs/stm32f1xx-hal #323: impl OutputPin with embedded-hal-1.0.0-alpha.4

pdgilbert, 2021-03-23:
Experimenting with @eldruin 's embedded-hal-1.0.0-alpha. ( https://github.com/eldruin/stm32f1xx-hal/tree/embedded-hal-1.0.0-alpha.4/) I am trying to get `impl OutputPin` to work with a `setup()` function. Using a modification of the `delay.rs `example in that fork I tried

```
//! "Blinky" using delays instead of a timer

#![deny(unsafe_code)]
#![no_main]
#![no_std]

use panic_halt as _;

use cortex_m_rt::entry;
use embedded_hal::digital::OutputPin;
use stm32f1xx_hal::{
    delay::Delay, pac, prelude::*,
    gpio::{gpioc::PC13, Output, PushPull},
};

use core::convert::Infallible;

fn setup() -> ( dyn OutputPin<Error = Infallible>, Delay ) {
//fn setup() -> (PC13<Output<PushPull>>, Delay ) {
    let dp = pac::Peripherals::take().unwrap();
    //let cp = cortex_m::Peripherals::take().unwrap();
    let cp = pac::CorePeripherals::take().unwrap();

    let mut flash = dp.FLASH.constrain();
    let mut rcc = dp.RCC.constrain();

    let clocks = rcc.cfgr.freeze(&mut flash.acr);

    let mut gpioc = dp.GPIOC.split(&mut rcc.apb2);

    #[cfg(feature = "stm32f100")]
    let led = gpioc.pc9.into_push_pull_output(&mut gpioc.crh);

    #[cfg(feature = "stm32f101")]
    let led = gpioc.pc9.into_push_pull_output(&mut gpioc.crh);

    #[cfg(any(feature = "stm32f103", feature = "stm32f105", feature = "stm32f107"))]
    let led = gpioc.pc13.into_push_pull_output(&mut gpioc.crh);

    let delay = Delay::new(cp.SYST, clocks);

    // return tuple  (led, delay)
    (
        led,
        delay,
    )
}

#[entry]
fn main() -> ! {
    let ( mut led, mut delay) = setup();

    loop {
        led.try_set_high().unwrap();
        delay.try_delay_ms(1_000_u16).unwrap();
        led.try_set_low().unwrap();
        delay.try_delay_ms(1_000_u16).unwrap();
    }
}
```
This builds with 

```
fn setup() -> (PC13<Output<PushPull>>, Delay ) {
```
but with
```
fn setup() -> ( dyn OutputPin<Error = Infallible>, Delay ) {
```
I get
```

error[E0277]: the size for values of type `(dyn OutputPin<Error = Infallible> + 'static)` cannot be known at compilation time
  --> examples/delay.rs:18:15
   |
18 | fn setup() -> ( dyn OutputPin<Error = Infallible>, Delay ) {
   |               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ doesn't have a size known at compile-time
   |
   = help: the trait `Sized` is not implemented for `(dyn OutputPin<Error = Infallible> + 'static)`
   = note: only the last element of a tuple may have a dynamically sized type
```
Reversing the order of the return tuple does not help, `Delay` does not have a known size either.

Is there a way to use impl trait for this?

TheZoq2, 2021-04-06:
I don't think you can return any `dyn` type because as the message suggests, the size of a `dyn` variable is not known. Would it work if you change `dyn` to `impl` here?

pdgilbert, 2021-04-08:
Hmmm, must have been too late at night when I banged my head on that. I <i>thought</i> I had started with `impl` and the compiler had directed me to `dyn`. But `impl` does work now.  Thanks.

closed 2021-04-08 by pdgilbert