stm32-rs/stm32f1xx-hal #334: UARTs should not enable DMA by default

Windfisch, 2021-05-12:
In https://github.com/stm32-rs/stm32f1xx-hal/blob/5d55f83528a1a4eebcc672201751a320d9be5854/src/serial.rs#L312, every U(S)ART enables the DMA transmit and receive bits (dmat and dmar) unconditionally when initializing the UART struct.

This interferes with other users of the DMA channels, and it appears to me that this should rather be done in `TransferPayload::start()/::stop()`: https://github.com/stm32-rs/stm32f1xx-hal/blob/5d55f83528a1a4eebcc672201751a320d9be5854/src/serial.rs#L530-L546), like in the SPI implementation https://github.com/stm32-rs/stm32f1xx-hal/blob/5d55f83528a1a4eebcc672201751a320d9be5854/src/spi.rs#L569-L586

Or did I overlook something here?

TheZoq2, 2021-05-17:
That certainly looks like a bug, thanks for reporting it! I'll look into fixing it when I find the time

burrbull, 2021-05-18:
I think `with_dma` and `split` are better places to move this. Not `start` and `stop` transfer.

Windfisch, 2021-05-19:
thanks @burrbull, this is indeed correct; I've updated my UART-pull request and would update the SPI-PR when I got some feedback on this one.
Could you review it please :)?

Windfisch, 2021-05-19:
Hm, I sort of dislike the name `split` for this function; do you think there's a strong reason to use that name? SPI does not have yet a `split` function that releases its DMA structs. I am considering calling that function differently, like `release`. What do you think?

burrbull, 2021-05-19:
> Hm, I sort of dislike the name `split` for this function; do you think there's a strong reason to use that name? SPI does not have yet a `split` function that releases its DMA structs. I am considering calling that function differently, like `release`. What do you think?

I think this is good idea.

Windfisch, 2021-05-19:
Do we have a policy when renaming functions or introducing other breaking changes?
- Do we just document them in the CHANGELOG?
- Do we offer the (deprecated) old function name along with the new name for some time until all users have notices and updated?
- Or do we just break compatibility?

Windfisch, 2021-05-20:
done, I updated both PRs. I think they are ready to merge now, if review don't say otherwise. Thanks for the feedback! :)

PR #336 (2021-07-13): Do not enable UART DMA flags unconditionally
Fixes #334

I have not tested that code, though, so proceed with caution :)
commit 8a758258 (2021-07-13): Do not enable UART DMA flags unconditionally (#336)

* Do not enable UART DMA flags unconditionally

Fixes #334

* More consistent use of release instead of free/split functions

* Fix warnings about mut arguments
closed 2021-07-13 by TeXitoi