stm32-rs/stm32f1xx-hal #350: PWM DMA support

fishrockz, 2021-07-28:
I notice that this repo dose not allow for DMA transfers to the TimX devices.

I recently followed https://stackoverflow.com/questions/63016570/how-to-make-dma-work-for-changing-the-duty-cycle-of-a-pwm-port-using-rust and i seem to have some thing very similar to that example working but for stm32f1, I have tested on a blue pill and get some cool wave forms out.

I will keep working on my example to create wave forms for my use case and will be trying to significantly reduce the unsafe code.

I would like to write this up, ideally as a addition to this lib. but I'm not sure what API to expose to users. I have not seen any other repos in https://github.com/stm32-rs that have DMA for there pwm.rs files other wise i would have just borrowed from one of them.

Is there any resources I could lean on when creating a PR or is there a reason that this functionality dose not already exist?

I notice that spi and other things use DMA but there API's are quite different so have struggled to use them as inspiration but at this point they are the best thing i have.

Many thanks

burrbull, 2021-07-28:
I've not heard about such implementation.
But you can ask in matrix channel: https://app.element.io/#/room/#rust-embedded:matrix.org

alexmadeathing, 2024-03-01:
Did you make any progress with this? I also need DMA to PWM output.

fishrockz, 2024-03-01:
Oh, yer I have quite a nice little motor controler for the one shot protical but I never made any progress in upstreaming it.

alexmadeathing, 2024-03-01:
I'm hoping to DMA to PWM duty cycle so that I can drive WS2812 asynchronously since they're vey time sensitive (synchronous timing is getting messed up by my other interrupts). Does that sound like the same sort of situation you overcame with your motors?

Have you got some code I can use as a reference? No worries if not!

fishrockz, 2024-03-03:
I think this is the main bit to get the DMA going for PWM

note that i was creating a pulse chain ie, set it off and then have several different pulses all spat out one after each other

```rust
    let c41 = gpiob.pb6.into_alternate_push_pull(&mut gpiob.crl);
    let c42 = gpiob.pb7.into_alternate_push_pull(&mut gpiob.crl);
    let c43 = gpiob.pb8.into_alternate_push_pull(&mut gpiob.crh);
    let c44 = gpiob.pb9.into_alternate_push_pull(&mut gpiob.crh);
    // If you don't want to use all channels, just leave some out
    // let c4 = gpioa.pa3.into_alternate_push_pull(&mut gpioa.crl);
    let pins4 = (c41, c42, c43, c44);

    let mut pwm4 =
        Timer::tim4(dp.TIM4, &clocks).pwm::<Tim4NoRemap, _, _, _>(pins4, &mut afio.mapr, 50.hz());

    let max4 = pwm4.get_max_duty();

    pwm4.set_duty(Channel::C1, 0);
    pwm4.enable(Channel::C1);

    // TIM2
    let c1 = gpioa.pa0.into_alternate_push_pull(&mut gpioa.crl);
    let c2 = gpioa.pa1.into_alternate_push_pull(&mut gpioa.crl);
    let c3 = gpioa.pa2.into_alternate_push_pull(&mut gpioa.crl);
    let c4 = gpioa.pa3.into_alternate_push_pull(&mut gpioa.crl);
    // If you don't want to use all channels, just leave some out
    // let c4 = gpioa.pa3.into_alternate_push_pull(&mut gpioa.crl);
    let pins = (c1, c2, c3, c4);

    let mut pwm = Timer::tim2(dp.TIM2, &clocks).pwm::<Tim2NoRemap, _, _, _>(
        pins,
        &mut afio.mapr,
        //1_250.khz(),
        625.khz(),
    );

    let max = pwm.get_max_duty();
    hprintln!("max {}", max).unwrap();

    pwm.set_duty(Channel::C1, 0);
    pwm.enable(Channel::C1);

    let tim2 = unsafe { &(*pac::TIM2::ptr()) }; // general timer 2 pointer
    let rcc_ptr = unsafe { &*pac::RCC::ptr() }; // RCC pointer
    let dma1 = unsafe { &*pac::DMA1::ptr() }; // DMA 1 pointer

    rcc_ptr.ahbenr.modify(|_, w| w.dma1en().set_bit()); // enable DMA1 clock: peripheral clock enable register

    // timer for DMA configuration
    tim2.dier.write(|w| w.tde().set_bit()); // enable DMA trigger
    tim2.dier.write(|w| w.ude().set_bit()); // enable update DMA request

    let _a = &tim2.ccr1 as *const _ as u32; // very different from 0x4000_0034

    // DMA configuration
    unsafe {
        dma1.ch2.par.write(|w| {
            w.pa()
                .bits(unsafe { &(*pac::TIM2::ptr()).dcr } as *const _ as u32)
        });
    };

    //dma1.cselr.write(|w| w.c2s().bits(0b0100)); // set CxS[3:0] to 0100 to map the DMA request to timer 2 channel 1
    //dma1.cselr.write(|w| w.c2s().bits(0b0100)); // set CxS[3:0] to 0100 to map the DMA request to timer 2 channel 1
    //dma1.cpar2.write(|w| w.pa().bits(0x4000_0034)); // set the DMA peripheral address register to the capture/compare 1 of TIM2
    unsafe {
        dma1.ch2.par.write(|w| w.pa().bits(0x4000_0034));
    }
    //dma1.cmar2.write(|w| w.ma().bits(buffer.as_ptr() as u32)); // write the buffer to the memory adress
[... 73 lines cut ...]

    let clean_command = new_command;
    let mut data: u16 = ((clean_command & 0b11111111111) << 5) | ((telm & 0b1) << 4);
    let mut checksum = (data >> 4) ^ (data >> 8) ^ (data >> 12);
    //if (telm & 0b1) == 1 {
    //    checksum = !checksum;
    //}
    checksum = checksum & 0x0f;
    //}
    data = data | checksum;

    for iii in 0..16 {
        let val: u16 = data & (1 << (15 - iii));
        if val != 0 {
            buffer[iii] = one_duty;
        } else {
            buffer[iii] = zero_duty;
        }
    }
}
```

I think a later version of this was a bit less unsafe but you do have to be careful about memory and lock and avoid dead locks etc

you have to be careful cos the interrupt could go off at any point in you main loop so you generally need something to co ordinate, can be as simple as turning off interrupts when you  update GLOBAL_BUFF or what ever you use handle state shared between the interrupt and the main loop

alexmadeathing, 2024-03-04:
Wow, you went above and beyond here. Thanks! Yes, I'll be using a pulse train too. I think my requirements are very similar. I'm using RTIC, so I'll just lean on that for resource sharing and interrupt free zones.

Why do you set TIM4 to 50hz early on? Is that just so you can get access to the PWM channel and then DMA with TIM2 takes over?

I'll leave you be now. You've given me more than enough to chew on. Thanks again.