stm32-rs/stm32f1xx-hal #356: Can not use SPI3

rfael, 2021-08-04:
Hi,
I am trying to use SPI3 on STM32F107 MCU, but I can not due to this error:
```
error[E0271]: type mismatch resolving `<Spi1Remap as stm32f1xx_hal::spi::sealed::Remap>::Periph == stm32f1xx_hal::pac::SPI3`
   --> src/main.rs:128:19
    |
128 |         let spi = stm32f1xx_hal::spi::Spi::spi3(
    |                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected struct `stm32f1xx_hal::pac::SPI3`, found struct `stm32f1xx_hal::pac::SPI1`
    |
    = note: required by `Spi::<stm32f1xx_hal::pac::SPI3, REMAP, PINS, u8>::spi3`
```

I think the compiler forces me to use remaped SPI1 instead of SPI3. In my case I can do that, but what if I want to use SPI1 and SPI3 at the same time?

Code snippet that I use to initialize the SPI3:
```Rust
        let (_pa15, pb3, pb4) = afio.mapr.disable_jtag(gpioa.pa15, gpiob.pb3, gpiob.pb4);

        let spi_pins = (
            pb3.into_alternate_push_pull(&mut gpiob.crl),
            pb4.into_floating_input(&mut gpiob.crl),
            gpiob.pb5.into_alternate_push_pull(&mut gpiob.crl),
        );

        let spi_mode = embedded_hal::spi::Mode {
            polarity: embedded_hal::spi::Polarity::IdleLow,
            phase: embedded_hal::spi::Phase::CaptureOnFirstTransition,
        };

        let spi = stm32f1xx_hal::spi::Spi::spi3(
            device.SPI3,
            spi_pins,
            &mut afio.mapr,
            spi_mode,
            100.khz(),
            clocks,
            &mut rcc_apb1,
        );
```

burrbull, 2021-08-04:
Try master branch.
Should be fixed by https://github.com/stm32-rs/stm32f1xx-hal/pull/312 .

tfx2001, 2021-08-11:
It has been fixed by #312. You can use below in your `Cargo.toml`
```toml
stm32f1xx-hal = { git = "https://github.com/stm32-rs/stm32f1xx-hal.git", features = ["stm32f103", "rt", "high"] }
```

closed 2021-08-11 by rfael