stm32-rs/stm32f1xx-hal #367: USB doesn't work on stm32f103

qthree, 2021-11-05:
I've tried standard `usb_serial` example on 3 different boards: one stm32f3discovery (with `stm32f3xx-hal` crate) and two different stm32f103 boards (one is bluepill-like `stm32f103c8t6` and another is full size noname board with `stm32f103vet6`).
Example works with stm32f3discovery but doesn't work with both stm32f103 boards. After plugging in USB `dmesg` shows bunch of `device descriptor read/64, error -32` and `Device not responding to setup address.`.
I've almost zero embedded experience, so I don't know what's going on here. But I've tried to flash [some random non-rust firmware](https://github.com/r2axz/bluepill-serial-monster) that uses USB and it worked.
I was using latest master, cause crates' version won't compile.

qthree, 2021-11-06:
This bug was introduced in #337, commit 67bdf68db80df7619f7deb42a878801ae1bfcdad works fine.

burrbull, 2021-11-06:
Looks like I missed something when rebase.

qthree, 2021-11-06:
Fixed by #368

closed 2021-11-06 by qthree