stm32-rs/stm32f1xx-hal #370: Could not find `can in stm32fxx_hal

joelimgu, 2021-11-11:
I am trying to use can and it seems to not be able to find it in the crate. I am using the ```version = "0.7.0"``` with ```features = ["stm32f103", "rt", "medium"]```

Any idea of what's happening, the files seems to be in the repo but cargo doesn't download it.
Thanks!

TheZoq2, 2021-11-15:
It looks like CAN wasn't added until after 0.7 was released, does it work better if you depend on the git version?

RCasatta, 2021-11-20:
I am patching `stm32f1xx-hal` also for using CAN, any release scheduled since it's more than 1 year there is no release?

joelimgu, 2021-11-21:
Yes, making it depend on the git version fixed it. Thanks! (we still had some problems with some dependencies but easy to solve with the compiler messages. But I still think is a bit confusing having the github version as 0.7 and the crates.io version also as 0.7 and not having the same content. just to point it out.

closed 2021-11-21 by joelimgu

burrbull, 2021-11-21:
> But I still think is a bit confusing having the github version as 0.7 and the crates.io version also as 0.7 and not having the same content. just to point it out.

For Rust project it is normal to grow version same day when it is published on crates.io

TheZoq2, 2021-11-21:
> I am patching stm32f1xx-hal also for using CAN, any release scheduled since it's more than 1 year there is no release?

I think we should do a release soon, I very much agree that 1 year since the last one is a bit long. I might not be able to do it right now, but emind me in a few weeks if that hasn't been done