stm32-rs/stm32f1xx-hal #374: `stm32f1xx_hal::pac::Interrupt: Nr` is not satisfied

zrz4066, 2021-11-18:
I can't compile OK after updating  stm32f1xx-hal.How to solve the issue. Thanks.

[image]

the trait bound `stm32f1xx_hal::pac::Interrupt: Nr` is not satisfied
the trait `Nr` is not implemented for `stm32f1xx_hal::pac::Interrupt`rustcE0277
nvic.rs(125, 12): required by a bound in `nvic::<impl cortex_m::peripheral::NVIC>::unmask`
error: the trait bound `stm32f1xx_hal::pac::Interrupt: Nr` is not [satisfied]([url]([url](url)))
label: the trait `Nr` is not implemented for `stm32f1xx_hal::pac::Interrupt`
note: required by a bound in `nvic::<impl cortex_m::peripheral::NVIC>::unmask`
label: the trait `Nr` is not implemented for `stm32f1xx_hal::pac::Interrupt`

burrbull, 2021-11-18:
Check `cortex-m` and `cortex-m-rt` version. Are they compatible with `stm32f1` 0.14?

zrz4066, 2021-11-18:
Thanks。     
 I change  from 
                 cortex_m::peripheral::NVIC::unmask(stm32f1xx_hal::pac::Interrupt::USART1);
       to
                 pac::NVIC::unmask(stm32f1xx_hal::pac::Interrupt::USART1);
    It compile ok
[image]

closed 2021-11-18 by zrz4066