stm32-rs/stm32f1xx-hal #376: Complementary PWM output?

alianse777, 2021-12-04:
Is there any way to configure complementary CHxN outputs, like CH1 CH1N?

thomask77, 2023-07-30:
I also have the same question.

Without this feature, stm32f1xx-hal is not suited for motor control projects :/

burrbull, 2023-07-31:
You could port main part of code from https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/timer/pwm.rs
But I yet don't have good idea how to workaround pin remapping.

thomask77, 2023-07-31:
I think I'll just go with low-level PAC access.
I also need to set up Center-Aligned-PWM and synchronize ADC, DMA channels and IRQs to the PWM.

I already have a working configuration generated by CubeMX. Only need to translate to Rust.

thomask77, 2023-07-31:
P.S.: Is there a table to see which HW features are not/partially/fully supported by the HAL?

hawav, 2024-05-07:
I am also attempting to implement BLDC motor control using this library. Could you please support Complementary PWM output functionality? Thank you for your consideration.

steewbsd, 2025-03-24:
For anyone looking at this in the future, I managed to get it working on TIM1_CH1N somewhat like this:
```rust  
        // Manual PAC initialization of the PWM Complementary channel as a                                                                                                                
        // workaround for the lack of it on the HAL                                                                                                                                       
        dp                                                                                                                                                                         
            .TIM1                                                                                                                                                                         
            .ccmr1_output()                                                                                                                                                               
            .write(|w| unsafe { w.oc1m().bits(0b110).cc1s().bits(0b00) });                                                                                                                                                                                                                                                               
        // Activate Main Output Enable (MOE) for CC1N                                                                                                                                     
        dp                                                                                                                                                                          
            .TIM1                                                                                                                                                                         
            .bdtr()                                                                                                                                                                       
            .write(|w| w.moe().set_bit().ossr().clear_bit());
        // Activate complementary output (CC1NE) and disable primary output (CC1E)                                                                                                        
        c.device                                                                                                                                                                          
            .TIM1                                                                                                                                                                         
            .ccer()                                                                                                                                                                       
            .write(|w| w.cc1ne().set_bit().cc1e().clear_bit());  
```
You'll also have to init the GPIOB (for PB13 in my case) and TIM1 clocks as usual.