stm32-rs/stm32f1xx-hal #424: Make generic types of `gpio::Cr<_, _>` (CRL and CRH) public

VanillaBrooks, 2022-06-05:
I am trying to write some abstractions around some pins and I need type level information to `gpio::gpioX::Parts.crl` and `gpio::gpioX::Parts.crh`. 

Currently (as far as rustdoc & I can tell) these types are not public. This prevents me from writing something like this trivial example:

```rust
use stm32f1xx_hal::gpio::{self, Input, Floating, PullDown, gpioa};

use gpioa::CRL; // does not work

fn make_pulldown(
    pa0: gpioa::PA0<Input<Floating>>,
    crh: &mut gpio::Cr<CRH, 'A'>,
    ) -> gpioa::PA0<Input<gpio::PullDown>> {
    pa0.into_pull_down_input(crh)
}
``` 

this will error with something like:

```

error[E0603]: struct `CRL` is private
    --> src/main.rs:4:12
     |
4    | use gpioa::CRL;
     |            ^^^ private struct
```

Is there a way to get around this, or perhaps a reason that these types are hidden?

Thanks!

burrbull, 2022-06-05:
make PR, please

VanillaBrooks, 2022-06-05:
Went to make the PR and i realized the current master branch uses const generics for `Cr` instead of the private type:

```
pub struct Cr<const P: char, const H: bool>(());
```

so this problem will be fixed on the next release and i'm closing this

closed 2022-06-05 by VanillaBrooks