stm32-rs/stm32f1xx-hal #425: WithPwm trait should be public

Fomys, 2022-07-24:
I writing a generic struct that takes a pwm channel to control a motor.

We've tried this code:
```rust
struct Motor<TIM: TimerExt, const C: u8> {
    pwm: PwmChannel<TIM, C>,
}
```
And I get the following error:
```rust
error[E0599]: the method `get_max_duty` exists for struct `PwmChannel<TIM, C>`, but its trait bounds were not satisfied
  --> base-roulante/src/main.rs:64:73
   |
64 |         let consigne = pid_correction / Self::MAX_CORRECTION * self.pwm.get_max_duty();
   |                                                                         ^^^^^^^^^^^^ method cannot be called on `PwmChannel<TIM, C>` due to unsatisfied trait bounds
   |
  ::: /home/fomys/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f1xx-hal-0.9.0/src/timer/pwm.rs:89:1
   |
89 | pub struct PwmChannel<TIM, const C: u8> {
   | --------------------------------------- doesn't satisfy `PwmChannel<TIM, C>: PwmPin`
   |
   = note: the following trait bounds were not satisfied:
           `TIM: stm32f1xx_hal::timer::sealed::WithPwm`
           `TIM: stm32f1xx_hal::timer::sealed::WithPwm`
           which is required by `PwmChannel<TIM, C>: PwmPin`
```

The ideal resolution would be to add trait bound with `WithPwm` trait, but it's impossible as it's a sealed trait.

Shouldn't it be public in order write this trait bound ? If this isn't the case, how should I resolve ?

burrbull, 2022-07-24:
Yeah. Generic code with sealed trait is often problematic.
As fast fix you could use `get_max_duty` from `PwmPin`:
https://github.com/stm32-rs/stm32f1xx-hal/blob/f9b24f4d9bac7fc3c93764bd295125800944f53b/src/timer/hal_02.rs#L136

Fomys, 2022-07-31:
To use PwmPin I have to add the trait bound `PwmPin<Duty = u16>` as the Duty is also generic.
Should I make a PR to make WithPwm public ?

Tnze, 2025-01-18:
I think we should move methods from `Pwm` and `PwmHz` to a new trait. Thus we can wrap PWMs in struct without writing trait bound of the TIM.

```rust
trait PWM {
    fn enable() ...
    fn disable()....
    fn get_duty()....
    ....
}

impl<TIM, REMAP, P, PINS> PWM for PwmHz<TIM, REMAP, P, PINS> where ... {
...
}

impl<TIM, REMAP, P, PINS> PWM for Pwm<TIM, REMAP, P, PINS> where ... {
....
}
```