stm32-rs/stm32f1xx-hal #432: PB3, PB4 does not work correctly

sb-child, 2022-08-16:
Hi,
I am trying this [example](https://github.com/stm32-rs/stm32f1xx-hal/blob/master/examples/nojtag.rs), PA15 is working fine but PB3, PB4 seems not blinking.
I have tested on `STM32F103RBT6` and `GD32F103RET6` with same results.

My `Cargo.toml` (deps are latest):
```toml
[package]
name = "example"
version = "1.0.0"
edition = "2021"

[profile.release]
opt-level = 'z'
lto = true
debug = false

[dependencies]
stm32f1xx-hal = { version = "*", features = ["stm32f103", "rt"] }
cortex-m-rt = { version = "*", features = ["device"] }
cortex-m = "*"
panic-halt = "*"
micromath = "*"
ssd1306 = "*"
embedded-hal = "*"
embedded-graphics = "*"
nb = "*"
```

burrbull, 2022-08-16:
https://docs.rs/stm32f1xx-hal/0.9.0/stm32f1xx_hal/gpio/index.html#accessing-pa15-pb3-and-pb14

https://github.com/stm32-rs/stm32f1xx-hal/blob/master/examples/nojtag.rs

sb-child, 2022-08-16:
> https://github.com/stm32-rs/stm32f1xx-hal/blob/master/examples/nojtag.rs

sorry that this example still doesn't work. PA15 is working fine but PB3, PB4 not

sb-child, 2022-08-16:
UPDATE: `STM32F103C8T6` works fine, but `STM32F103RBT6` not.

burrbull, 2022-08-16:
Those chips should be identical inside except memory size, so I can't say what is wrong.