stm32-rs/stm32f1xx-hal #437: Clocks can be created trivially

horazont, 2022-08-27:
The [documentation of `crate::rcc::Clocks`](https://docs.rs/stm32f1xx-hal/latest/stm32f1xx_hal/rcc/struct.Clocks.html) claims:

> The existence of this value indicates that the clock configuration can no longer be changed

But the following example code can be used to obtain a value of `crate::rcc::Clocks` without actually tying the clocks down:

```rust
#![no_std]
#![no_main]

use panic_halt as _;

use stm32f1xx_hal as hal;

use cortex_m_rt::entry;

#[entry]
fn main() -> ! {
	let clocks: hal::rcc::Clocks = hal::rcc::Config{
		hse: Some(8000000),
		pllmul: Some(8),
		hpre: hal::rcc::HPre::DIV1,
		ppre1: hal::rcc::PPre::DIV2,
		ppre2: hal::rcc::PPre::DIV1,
		adcpre: hal::rcc::AdcPre::DIV8,
		usbpre: hal::rcc::UsbPre::DIV1,
	}.get_clocks();

	loop{}
}
```

burrbull, 2022-08-28:
https://docs.rs/stm32f1xx-hal/latest/stm32f1xx_hal/rcc/struct.Config.html#method.get_clocks should be private.
Thanks.
Could you make PR?

horazont, 2022-08-28:
Yep sure, done -> #438.

commit 4909f29c (2022-08-28): Un-publish crate::rcc::Config::get_clocks

The existence of a crate::rcc::Clocks value is supposed to assert that
the MCU clocks have been configured and that they cannot be changed
anymore. That works by consuming the only instance of CFGR in the
corresponding function and then returning a Clocks value.

However, having crate::rcc::config::Config::get_clocks public implies
that external code can, at any time, generate a Clocks value: by simply
constructing a Config instance (all fields are public) and calling the
function.

```rust
#![no_std]
#![no_main]

use panic_halt as _;

use stm32f1xx_h [... cut]
commit f063d5be (2022-08-28): Merge #438

438: Un-publish crate::rcc::Config::get_clocks r=burrbull a=horazont

The existence of a crate::rcc::Clocks value is supposed to assert that
the MCU clocks have been configured and that they cannot be changed
anymore. That works by consuming the only instance of CFGR in the
corresponding function and then returning a Clocks value.

However, having crate::rcc::config::Config::get_clocks public implies
that external code can, at any time, generate a Clocks value: by simply
constructing a Config instance (all fields are public) and calling the
function.

```rust
#![no_std]
#![no_main] [... cut]
PR #438 (2022-08-28): Un-publish crate::rcc::Config::get_clocks
The existence of a crate::rcc::Clocks value is supposed to assert that
the MCU clocks have been configured and that they cannot be changed
anymore. That works by consuming the only instance of CFGR in the
corresponding function and then returning a Clocks value.

However, having crate::rcc::config::Config::get_clocks public implies
that external code can, at any time, generate a Clocks value: by simply
constructing a Config instance (all fields are public) and calling the
function.

```rust
#![no_std]
#![no_main]

use panic_halt as _;

use stm32f1xx_ha [... cut]
closed 2022-08-28 by bors[bot]