stm32-rs/stm32f1xx-hal #447: Unable to build blinky

SilvanRehm, 2022-10-01:
I just started with rust, the first example I tried was https://jonathanklimt.de/electronics/programming/embedded-rust/rust-on-stm32-2/  which worked. Next I wanted to test CANBus, which I could not get running. So I tried the blinky as in the readme first. But it won't build. Here's the output:
(Details how I set up the project below)

```
error[E0061]: this function takes 1 argument but 0 arguments were supplied
  --> src/main.rs:36:30
   |
36 |     let mut gpioc = dp.GPIOC.split();
   |                              ^^^^^-- an argument of type `&mut APB2` is missing
   |
note: associated function defined here
  --> /home/silvan/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32f1xx-hal-0.6.1/src/gpio.rs:86:8
   |
86 |     fn split(self, apb2: &mut APB2) -> Self::Parts;
   |        ^^^^^
help: provide the argument
   |
36 |     let mut gpioc = dp.GPIOC.split(/* &mut APB2 */);
   |                              ~~~~~~~~~~~~~~~~~~~~~~

error[E0599]: no method named `counter_hz` found for struct `Timer` in the current scope
  --> src/main.rs:42:51
   |
42 |     let mut timer = Timer::syst(cp.SYST, &clocks).counter_hz();
   |                                                   ^^^^^^^^^^ method not found in `Timer<SYST>`

error[E0599]: no method named `Hz` found for type `{integer}` in the current scope
  --> src/main.rs:43:19
   |
43 |     timer.start(1.Hz()).unwrap();
   |                   ^^ method not found in `{integer}`

error[E0599]: no method named `set_high` found for struct `PC13` in the current scope
  --> src/main.rs:48:13
   |
48 |         led.set_high();
   |             ^^^^^^^^ method not found in `PC13<Output<PushPull>>`
   |
  ::: /home/silvan/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-hal-0.2.7/src/digital/v2.rs:60:8
   |
60 |     fn set_high(&mut self) -> Result<(), Self::Error>;
   |        -------- the method is available for `PC13<Output<PushPull>>` here
   |
   = help: items from traits can only be used if the trait is in scope
help: the following trait is implemented but not in scope; perhaps add a `use` for it:
   |
12 | use embedded_hal::digital::v2::OutputPin;
   |

error[E0599]: no method named `set_low` found for struct `PC13` in the current scope
  --> src/main.rs:50:13
   |
50 |         led.set_low();
   |             ^^^^^^^ method not found in `PC13<Output<PushPull>>`
   |
  ::: /home/silvan/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-hal-0.2.7/src/digital/v2.rs:54:8
   |
54 |     fn set_low(&mut self) -> Result<(), Self::Error>;
   |        ------- the method is available for `PC13<Output<PushPull>>` here
   |
   = help: items from traits can only be used if the trait is in scope
help: the following trait is implemented but not in scope; perhaps add a `use` for it:
   |
12 | use embedded_hal::digital::v2::OutputPin;
   |

Some errors have detailed explanations: E0061, E0599.
For more information about an error, try `rustc --explain E0061`.
error: could not compile `blinky3` due to 5 previous errors
```

Here's what I did: 

```
cargo init blinky3 
cd blinky3
```

paste
```
[target.thumbv7m-none-eabi]
runner = 'arm-none-eabi-gdb'
rustflags = [
  "-C", "link-arg=-Tlink.x",
]

[build]
target = "thumbv7m-none-eabi"
```
into .cargo/config

paste
```
/* Linker script for the STM32F103C8T6 */
MEMORY
{
  FLASH : ORIGIN = 0x08000000, LENGTH = 64K
  RAM : ORIGIN = 0x20000000, LENGTH = 20K
}
```
into memory.x

my Cargo.toml:
```
[package]
name = "blinky3"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]              
embedded-hal = "0.2.3"
nb = "0.1.2"
cortex-m = "0.6.2"    
cortex-m-rt = "0.6.11"
# Panic behaviour, see https://crates.io/keywords/panic-impl for alternatives
panic-halt = "0.2.0"

[dependencies.stm32f1xx-hal]
version = "0.6.1"
features = ["rt", "stm32f103", "medium"]
```

copy blinky.rs example into src/main.rs:
```
//! Blinks an LED
//!
//! This assumes that a LED is connected to pc13 as is the case on the blue pill board.
//!
//! Note: Without additional hardware, PC13 should not be used to drive an LED, see page 5.1.2 of
//! the reference manual for an explanation. This is not an issue on the blue pill.

#![deny(unsafe_code)]
#![no_std]
#![no_main]

use panic_halt as _;

use nb::block;

use cortex_m_rt::entry;
use stm32f1xx_hal::{pac, prelude::*, timer::Timer};

#[entry]
fn main() -> ! {
    // Get access to the core peripherals from the cortex-m crate
    let cp = cortex_m::Peripherals::take().unwrap();
    // Get access to the device specific peripherals from the peripheral access crate
    let dp = pac::Peripherals::take().unwrap();

    // Take ownership over the raw flash and rcc devices and convert them into the corresponding
    // HAL structs
    let mut flash = dp.FLASH.constrain();
    let rcc = dp.RCC.constrain();

    // Freeze the configuration of all the clocks in the system and store the frozen frequencies in
    // `clocks`
    let clocks = rcc.cfgr.freeze(&mut flash.acr);

    // Acquire the GPIOC peripheral
    let mut gpioc = dp.GPIOC.split();

    // Configure gpio C pin 13 as a push-pull output. The `crh` register is passed to the function
    // in order to configure the port. For pins 0-7, crl should be passed instead.
    let mut led = gpioc.pc13.into_push_pull_output(&mut gpioc.crh);
    // Configure the syst timer to trigger an update every second
    let mut timer = Timer::syst(cp.SYST, &clocks).counter_hz();
    timer.start(1.Hz()).unwrap();

    // Wait for the timer to trigger an update and change the state of the LED
    loop {
        block!(timer.wait()).unwrap();
        led.set_high();
        block!(timer.wait()).unwrap();
        led.set_low();
    }
}
```

zacck, 2022-10-05:
When splittign GPIO you do need to pass through `&mut rcc.apb2` I think its a new parameter of the API

Njolf, 2023-01-08:
Hey! I had this same issue. the dependencies in ` cargo.toml ` are out of date. I fixed mine by just going through the dependency crates and manually checking their latest version on crates.io and updating them. I think just the ` cortex-m ` and ` cortex-m-rt ` are necessary in order for it to compile, but here's the entire dependency list from my example, at the time of writing:
` [dependencies] `
` embedded-hal = "0.2.7" `
` nb = "1.0.0" `
` cortex-m = "0.7.6" `
` cortex-m-rt = "0.7.2" `
` panic-halt = "0.2.0" `
Hope this helps!

lallbj23, 2023-07-17:
I used this method to solve it.
Attach the dependencies in my file.

`
[dependencies]
embedded-hal = "0.2.7"
nb = "1.1.0"
cortex-m = "0.7.7"
cortex-m-rt = "0.7.3"
panic-halt = "0.2.0"
[dependencies.stm32f1xx-hal]
version = "0.10.0"
features = ["rt", "stm32f103", "medium"]
`