stm32-rs/stm32f1xx-hal #472: Problem with building

hanusek, 2023-09-07:
Hello.
I have a problem with building this library.

```
git clone ... 
git checkout v0.10.0 
cargo c
```

**Error log:**
```
mhanusek@et-0989:~/work/code/praktyka2023/stm32f1xx-hal$ git checkout v0.10.0 
M	.cargo/config
HEAD is now at 21c64d8 Merge #452
mhanusek@et-0989:~/work/code/praktyka2023/stm32f1xx-hal$ RUST_BACKTRACE=1 cargo c
   Compiling quote v1.0.33
   Compiling bare-metal v0.2.5
    Checking bxcan v0.7.0
   Compiling stm32f1 v0.15.1
    Checking fugit-timer v0.1.3
    Checking cortex-m v0.7.7
error: failed to run custom build command for `stm32f1 v0.15.1`

Caused by:
  process didn't exit successfully: `/home/mhanusek/work/code/praktyka2023/stm32f1xx-hal/target/debug/build/stm32f1-9b3b17175b6811de/build-script-build` (exit status: 101)
  --- stdout
  cargo:rustc-link-search=/home/mhanusek/work/code/praktyka2023/stm32f1xx-hal/target/thumbv7m-none-eabi/debug/build/stm32f1-7fb1984510008b36/out

  --- stderr
  thread 'main' panicked at 'No device features selected', /home/mhanusek/.cargo/registry/src/index.crates.io-6f17d22bba15001f/stm32f1-0.15.1/build.rs:18:18
  stack backtrace:
     0: std::panicking::begin_panic
               at /rustc/5680fa18feaa87f3ff04063800aec256c3d4b4be/library/std/src/panicking.rs:625:12
     1: build_script_build::main
               at ./build.rs:18:18
     2: core::ops::function::FnOnce::call_once
               at /rustc/5680fa18feaa87f3ff04063800aec256c3d4b4be/library/core/src/ops/function.rs:250:5
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
warning: build failed, waiting for other jobs to finish...
```

My file `.cargo/config`:
```
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = "probe-run --chip STM32F103VE"

[build]
# Always compile for the instruction set of the STM32F1
target = "thumbv7m-none-eabi"

# use the Tlink.x scrip from the cortex-m-rt crate
rustflags = [
"-C", "link-arg=-Tlink.x",
"-C", "link-arg=--nmagic",
"-C", "link-arg=-Tdefmt.x",
]
```

**toolchains:**
```
rustup show
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/mhanusek/.rustup

installed toolchains
--------------------

stable-x86_64-unknown-linux-gnu
nightly-x86_64-unknown-linux-gnu
1.49-x86_64-unknown-linux-gnu
1.54-x86_64-unknown-linux-gnu
1.59-x86_64-unknown-linux-gnu
1.49.0-x86_64-unknown-linux-gnu
1.54.0-x86_64-unknown-linux-gnu
1.58.0-x86_64-unknown-linux-gnu
1.58.1-x86_64-unknown-linux-gnu
1.62.1-x86_64-unknown-linux-gnu
1.64.0-x86_64-unknown-linux-gnu
1.65.0-x86_64-unknown-linux-gnu
1.66.0-arm-unknown-linux-gnueabihf
1.66.0-x86_64-unknown-linux-gnu
1.68.0-x86_64-unknown-linux-gnu
1.70.0-x86_64-unknown-linux-gnu
1.72.0-x86_64-unknown-linux-gnu (default)

installed targets for active toolchain
--------------------------------------

thumbv7m-none-eabi
x86_64-unknown-linux-gnu

active toolchain
----------------

1.72.0-x86_64-unknown-linux-gnu (default)
rustc 1.72.0 (5680fa18f 2023-08-23)
```

TeXitoi, 2023-09-07:
Let's highlight one line:

```
  thread 'main' panicked at 'No device features selected', /home/mhanusek/.cargo/registry/src/index.crates.io-6f17d22bba15001f/stm32f1-0.15.1/build.rs:18:18
```

hanusek, 2023-09-07:
@TeXitoi thanks for your reply
next problem:
```
cargo b --features="stm32f103" --features="rt" --features="high" --example blinky_timer_irq

probe-run --chip STM32F103VE ./target/thumbv7m-none-eabi/debug/examples/blinky_timer_irq
(HOST) INFO  flashing program (22 pages / 22.00 KiB)
(HOST) INFO  success!
Error: `.defmt` section found, but no version symbol - check your linker configuration
```

burrbull, 2023-09-07:
```
"-C", "link-arg=-Tdefmt.x",
```
comment this line in `.cargo/config`

closed 2023-09-22 by hanusek