stm32-rs/stm32f1xx-hal #486: compilation time error

Bariskizilkaya, 2024-05-15:
I followed untill here
[image]

but  i got different kind of error

[image]

I m also new in rust could you please help me?

TheZoq2, 2024-05-15:
I think you can ignore that and keep following the guide. This error looks like a better error than what was there when we wrote the guide.

If you move on and run into other issues, let us know

Bariskizilkaya, 2024-05-15:
when i continue the guide i created 2 more files which are memory.x .config/config files and i got this linking problem
[image]