stm32-rs/stm32f1xx-hal #490: The trait `UsbPeripheral` is not implemented for `usb::Peripheral`

justapig9020, 2024-07-26:
**Describe**
The  [doc](https://docs.rs/stm32f1xx-hal/0.10.0/stm32f1xx_hal/usb/struct.Peripheral.html) says `usb::Peripheral` has implemented [UsbPeripheral](https://docs.rs/stm32-usbd/0.6.0/x86_64-unknown-linux-gnu/stm32_usbd/trait.UsbPeripheral.html). But I got a compile error says the struct has not implement the trait.

**To Reproduce**
- main.rs
```rust
#![no_std]
#![no_main]

use cortex_m_rt::entry;
use stm32f1xx_hal::{self as hal, gpio::GpioExt};
use panic_rtt_target as _;

#[entry]
fn main() -> ! {
    let dp = hal::pac::Peripherals::take().unwrap();
    let mut gpioa = dp.GPIOA.split();
    let usb = hal::usb::Peripheral {
        usb: dp.USB,
        pin_dm: gpioa.pa11,
        pin_dp: gpioa.pa12.into_floating_input(&mut gpioa.crh),
    };
    let usb_bus = stm32_usbd::UsbBus::new(usb); // compile error here
    loop {
    }
}
```

- Cargo.toml
```
[package]
name = "reproduce"
version = "0.1.0"
edition = "2021"

[dependencies]
cortex-m = "0.7.7"
cortex-m-rt = "0.7.3"
panic-semihosting = "0.6.0"
usb-device = "0.3.2"
stm32-usbd = "0.7.0"
panic-rtt-target = "0.1.3"
rtt-target = "0.5.0"

[dependencies.stm32f1xx-hal]
version = "0.10.0"
features = ["rt", "stm32f103", "medium", "stm32-usbd"]
```

- memory.x
```
/* Linker script for the STM32F103C8T6 */
MEMORY
{
  FLASH : ORIGIN = 0x08000000, LENGTH = 64K
  RAM : ORIGIN = 0x20000000, LENGTH = 20K
}
```

- .cargo/config.toml
```toml
[target.thumbv7m-none-eabi]
runner = 'arm-none-eabi-gdb'
rustflags = [
  "-C", "link-arg=-Tlink.x",
]

[build]
target = "thumbv7m-none-eabi"
```

**Expected behavior**
`usb::Peripheral` implements [UsbPeripheral](https://docs.rs/stm32-usbd/0.6.0/x86_64-unknown-linux-gnu/stm32_usbd/trait.UsbPeripheral.html) trait and is able to construct the [UsbBusAllocator](https://docs.rs/usb-device/0.2.7/usb_device/bus/struct.UsbBusAllocator.html).

**Error mesage**
```
$ cargo build --release
    Compiling keyboard-polling v0.1.0 (/Users/jup/learn/rust-embedded/keyboard-polling)
error[E0277]: the trait bound `Peripheral: UsbPeripheral` is not satisfied
   --> src/main.rs:107:43
    |
107 |     let usb_bus = stm32_usbd::UsbBus::new(usb);
    |                   ----------------------- ^^^ the trait `UsbPeripheral` is not implemented for `Peripheral`
    |                   |
    |                   required by a bound introduced by this call
    |
note: required by a bound in `stm32_usbd::UsbBus::<USB>::new`
   --> /Users/jup/.cargo/registry/src/index.crates.io-6f17d22bba15001f/stm32-usbd-0.7.0/src/bus.rs:23:11
    |
23  | impl<USB: UsbPeripheral> UsbBus<USB> {
    |           ^^^^^^^^^^^^^ required by this bound in `UsbBus::<USB>::new`
24  |     /// Constructs a new USB peripheral driver.
25  |     pub fn new(peripheral: USB) -> UsbBusAllocator<Self> {
    |            --- required by a bound in this associated function

error[E0277]: the trait bound `Peripheral: UsbPeripheral` is not satisfied
   --> src/main.rs:107:19
    |
107 |     let usb_bus = stm32_usbd::UsbBus::new(usb);
    |                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `UsbPeripheral` is not implemented for `Peripheral`, which is required by `stm32_usbd::UsbBus<Peripheral>: usb_device::bus::UsbBus`
    |
    = help: the trait `usb_device::bus::UsbBus` is implemented for `stm32_usbd::UsbBus<USB>`
    = note: required for `stm32_usbd::UsbBus<Peripheral>` to implement `usb_device::bus::UsbBus`
note: required by a bound in `usb_device::bus::UsbBusAllocator`
   --> /Users/jup/.cargo/registry/src/index.crates.io-6f17d22bba15001f/usb-device-0.3.2/src/bus.rs:152:31
    |
152 | pub struct UsbBusAllocator<B: UsbBus> {
    |                               ^^^^^^ required by this bound in `UsbBusAllocator`

For more information about this error, try `rustc --explain E0277`.
error: could not compile `keyboard-polling` (bin "keyboard-polling") due to 2 previous errors

```

**Environment: **
 - OS: MacOS
 - Rustc: rustc 1.81.0-nightly (24d2ac0b5 2024-07-15)
 - Cargo: cargo 1.81.0-nightly (154fdac39 2024-07-07)
 - Rustup: rustup 1.27.1 (54dd3d00f 2024-04-24)

RarePossum, 2024-08-27:
I'm having the same issue with a different `Peripheral`. My code is:

```rust
use cortex_m::asm::delay;
use cortex_m_rt::entry;
use stm32f1xx_hal::usb::{Peripheral, UsbBus};
use stm32f1xx_hal::{pac, prelude::*};
use usb_device::prelude::*;
use usbd_serial::{SerialPort, USB_CLASS_CDC};

#[entry]
fn main() -> ! {
    let dp = pac::Peripherals::take().unwrap();
    let mut gpioa = dp.GPIOA.split();
    let mut usb_dp = gpioa.pa12.into_push_pull_output(&mut gpioa.crh);
    usb_dp.set_low();
    delay(clocks.sysclk().raw() / 100);

    let usb = stm32f1xx_hal::usb::Peripheral {
        usb: dp.USB,
        pin_dm: gpioa.pa11,
        pin_dp: usb_dp.into_floating_input(&mut gpioa.crh),
    };
    let usb_bus = UsbBus::new(usb);
    let mut serial = SerialPort::new(&usb_bus);
}
```
The error produced is:

```
error[E0308]: mismatched types
   --> src/main.rs:59:38
    |
59  |     let mut serial = SerialPort::new(&usb_bus);
    |                      --------------- ^^^^^^^^ expected `&UsbBusAllocator<_>`, found `&UsbBusAllocator<UsbBus<...>>`
    |                      |
    |                      arguments to this function are incorrect
    |
    = note: `UsbBusAllocator<UsbBus<...>>` and `UsbBusAllocator<_>` have similar names, but are actually distinct types
note: `UsbBusAllocator<UsbBus<...>>` is defined in crate `usb_device`
```

For reference my `cargo.toml` is
```toml
[package]
name = "blinky-test"
version = "0.1.0"
edition = "2021"

[dependencies]
embedded-hal = "1.0.0"
nb = "1"
cortex-m = "0.7.6"
cortex-m-rt = "0.7.1"
# Panic behaviour, see https://crates.io/keywords/panic-impl for alternatives
panic-halt = "0.2.0"
cortex-m-semihosting = "0.5.0"
usb-device = "0.3.2"
usbd-serial = "0.2.2"
stm32-usbd = "0.7.0"
stm32f1xx-hal = { features=["rt", "stm32f103", "medium", "stm32-usbd"], version =  "0.10.0" }
panic-semihosting = "0.6.0"
panic-itm = "0.4.2"
```

### Environment
- OS: Opensuse Tumbleweed
- Rustc: rustc 1.82.0-nightly (f167efad2 2024-08-24)
- cargo: cargo 1.82.0-nightly (8f40fc59f 2024-08-21)
- rustup: rustup 1.27.0 (2024-03-15)

RarePossum, 2024-08-27:
And rolling back to 
```toml
[dependencies]
usb-device = "0.2.8"
usbd-serial = "0.1.1"
```
fixes it. Alright then.