stm32-rs/stm32f1xx-hal #519: I2c with Alternate<OpenDrain>

pdgilbert, 2024-12-14:
(Using git master) I have code for setting `i2c1` which works with `pb6,pb7` in several device crates using `embedded-hal 1.0.0`. When I try to use `pb8,pb9` which need `Alternate<OpenDrain>`  I get the errors below, with code further below. I am not sure if this is a bug or if I have the wrong incantation for Alternate OpenDrain. Something seems a bit odd about the `0` in
```
note: required for `Pin<'B', 8, Alternate<OpenDrain>>` to implement `RInto<stm32f1xx_hal::afio::i2c1::Scl, 0>`
```
since I think it needs `gpiob.crh`.

<details>
<summary>Click to expand pb8, pb9 alternate_open_drain error messages</summary>

```
$ cargo build --no-default-features --target thumbv7m-none-eabi --features stm32f103,stm32f1xx  --example setup-i2c
   Compiling testing-example v0.0.1 (../test_itc_f1xx)
error[E0277]: the trait bound `stm32f1xx_hal::afio::i2c1::Scl: RFrom<stm32f1xx_hal::gpio::Pin<'B', 8, stm32f1xx_hal::gpio::Alternate<stm32f1xx_hal::gpio::OpenDrain>>, 0>` is not satisfied
   --> examples/setup-i2c.rs:116:50
    |
116 |     let i2c1 = BlockingI2c::<I2C1>::new(dp.I2C1, (scl, sda),
    |                ------------------------          ^^^^^^^^^^ unsatisfied trait bound
    |                |
    |                required by a bound introduced by this call
    |
    = help: the trait `RFrom<stm32f1xx_hal::gpio::Pin<'B', 8, stm32f1xx_hal::gpio::Alternate<stm32f1xx_hal::gpio::OpenDrain>>, 0>` is not implemented for `stm32f1xx_hal::afio::i2c1::Scl`
    = help: the following other types implement trait `RFrom<T, R>`:
              `stm32f1xx_hal::afio::i2c1::Scl` implements `RFrom<stm32f1xx_hal::gpio::Pin<'B', 6, stm32f1xx_hal::gpio::Alternate<stm32f1xx_hal::gpio::OpenDrain>>, 0>`
              `stm32f1xx_hal::afio::i2c1::Scl` implements `RFrom<stm32f1xx_hal::gpio::Pin<'B', 6>, 0>`
              `stm32f1xx_hal::afio::i2c1::Scl` implements `RFrom<stm32f1xx_hal::gpio::Pin<'B', 8, stm32f1xx_hal::gpio::Alternate<stm32f1xx_hal::gpio::OpenDrain>>, 1>`
              `stm32f1xx_hal::afio::i2c1::Scl` implements `RFrom<stm32f1xx_hal::gpio::Pin<'B', 8>, 1>`
    = note: required for `Pin<'B', 8, Alternate<OpenDrain>>` to implement `RInto<stm32f1xx_hal::afio::i2c1::Scl, 0>`
note: required by a bound in `BlockingI2c::<I2C>::new`
   --> /.cargo/git/checkouts/stm32f1xx-hal-bb9d214e810c7b47/f62c083/src/i2c/blocking.rs:25:21
    |
23  |     pub fn new(
    |            --- required by a bound in this associated function
24  |         i2c: I2C,
25  |         pins: (impl RInto<I2C::Scl, 0>, impl RInto<I2C::Sda, 0>),
    |                     ^^^^^^^^^^^^^^^^^^ required by this bound in `BlockingI2c::<I2C>::new`
    = note: the full name for the type has been written to '../test_itc_f1xx/target/thumbv7m-none-eabi/debug/examples/setup_i2c-7f777f03336668fc.long-type-6265293051827038659.txt'
    = note: consider using `--verbose` to print the full type name to the console

error[E0277]: the trait bound `stm32f1xx_hal::afio::i2c1::Sda: RFrom<stm32f1xx_hal::gpio::Pin<'B', 9, stm32f1xx_hal::gpio::Alternate<stm32f1xx_hal::gpio::OpenDrain>>, 0>` is not satisfied
   --> examples/setup-i2c.rs:116:50
    |
116 |     let i2c1 = BlockingI2c::<I2C1>::new(dp.I2C1, (scl, sda),
    |                ------------------------          ^^^^^^^^^^ unsatisfied trait bound
    |                |
    |                required by a bound introduced by this call
    |
    = help: the trait `RFrom<stm32f1xx_hal::gpio::Pin<'B', 9, stm32f1xx_hal::gpio::Alternate<stm32f1xx_hal::gpio::OpenDrain>>, 0>` is not implemented for `stm32f1xx_hal::afio::i2c1::Sda`
    = help: the following other types implement trait `RFrom<T, R>`:
              `stm32f1xx_hal::afio::i2c1::Sda` implements `RFrom<stm32f1xx_hal::gpio::Pin<'B', 7, stm32f1xx_hal::gpio::Alternate<stm32f1xx_hal::gpio::OpenDrain>>, 0>`
              `stm32f1xx_hal::afio::i2c1::Sda` implements `RFrom<stm32f1xx_hal::gpio::Pin<'B', 7>, 0>`
              `stm32f1xx_hal::afio::i2c1::Sda` implements `RFrom<stm32f1xx_hal::gpio::Pin<'B', 9, stm32f1xx_hal::gpio::Alternate<stm32f1xx_hal::gpio::OpenDrain>>, 1>`
              `stm32f1xx_hal::afio::i2c1::Sda` implements `RFrom<stm32f1xx_hal::gpio::Pin<'B', 9>, 1>`
    = note: required for `Pin<'B', 9, Alternate<OpenDrain>>` to implement `RInto<stm32f1xx_hal::afio::i2c1::Sda, 0>`
note: required by a bound in `BlockingI2c::<I2C>::new`
   --> /.cargo/git/checkouts/stm32f1xx-hal-bb9d214e810c7b47/f62c083/src/i2c/blocking.rs:25:46
    |
23  |     pub fn new(
    |            --- required by a bound in this associated function
24  |         i2c: I2C,
25  |         pins: (impl RInto<I2C::Scl, 0>, impl RInto<I2C::Sda, 0>),
    |                                              ^^^^^^^^^^^^^^^^^^ required by this bound in `BlockingI2c::<I2C>::new`
    = note: the full name for the type has been written to '../test_itc_f1xx/target/thumbv7m-none-eabi/debug/examples/setup_i2c-7f777f03336668fc.long-type-2758305641861797322.txt'
    = note: consider using `--verbose` to print the full type name to the console

For more information about this error, try `rustc --explain E0277`.
error: could not compile `testing-example` (example "setup-i2c") due to 2 previous errors

```
</details>

[... 132 lines cut ...]

#[entry]
fn main() -> ! {
    
    let (_i2c1, mut delay, mut delay2) = setup();

    hprintln!("Start the sensor...");
    //    example 1
    // let mut aht  = Aht20::new(i2c1, DEFAULT_I2C_ADDRESS, &mut delay).expect("sensor initialization failed.");

    delay.delay_ms(500);   // this is a DelayNs

    loop {
        hprintln!("aht.measure()");

        // example 1
        //let th = aht.measure().unwrap();   // Read humidity and temperature.
        //hprintln!("{:.3}C  {}% RH", th.temperature.celcius(), th.relative_humidity);

        delay2.delay(5000.millis());   // this is a SysDelay
    }
}

```
</details>

burrbull, 2024-12-14:
`remap` is explicit now. This form should work https://github.com/stm32-rs/stm32f1xx-hal/blob/f62c083655965b2d92e208d463499891fc9c1f76/examples/i2c-bme280/src/main.rs#L60-L70

`::new` allows only `Remap<0>` for now. I need to do more experiments to make it more useful.

burrbull, 2024-12-14:
Try #520. Just a draft for now:

```rust
    let i2c1 = BlockingI2c::new(dp.I2C1.remap(&mut afio.mapr), (scl, sda),
                   Mode::Fast {frequency: 400.kHz(), duty_cycle: DutyCycle::Ratio16to9,},
                   &clocks, 1000, 10, 1000, 1000,);
```

pdgilbert, 2024-12-14:
Thanks @burrbull and sorry for the bother. I read that comment about `remap` and forgot about it when I needed it. I got the    `i2c = dp ...`  version working with master branch and the `i2c1 = BlockingI2c::new(...` working with branch `rmp-new`.

If new examples are helpful I can submit as a PR? Would it be better to wait until you draft is done?

closed 2024-12-14 by pdgilbert

burrbull, 2024-12-14:
> If new examples are helpful I can submit as a PR? Would it be better to wait until you draft is done?

Prepare them. I'll look tomorrow.