stm32-rs/stm32f1xx-hal #523: i2c runtime error ... todo!();

pdgilbert, 2025-01-01:
On bluepill I am checking the examples as in https://github.com/stm32-rs/stm32f4xx-hal/issues/828. Example  `sht30-em-semi` gives a "not yet implemented"  runtime error at
```
Remote target In: sht30_em_semi::__cortex_m_rt_main         L301  PC: 0x80003b6
at .cargo/git/checkouts/stm32f1xx-hal-bb9d214e810c7b47/6c5dc88/src/i2c/block
ing.rs:301
301          todo!();        
```
This is in  fn `transaction_slice`

(Examples `aht20-bl-semi`, `aht20-dr-semi`, `aht20-em-semi`, `shtc3-semi`, ` xca9548a-aht20-bl`, `xca9548a-aht20-dr`, and `xca9548a-aht20-em` all work.)

burrbull, 2025-01-01:
See https://github.com/stm32-rs/stm32f4xx-hal/blob/585dd0f30443c45a1be9f6483fd480b141d819a2/src/i2c.rs#L627 for  implementation example.

tmehr-enqt, 2025-09-10:
Implemented with version 0.11

closed 2025-09-10 by tmehr-enqt