stm32-rs/stm32f1xx-hal #525: Please publish new release to crates.io

Tnze, 2025-01-18:
It's 2 years later since the latest version has released.

liangzhaoyuan12, 2025-08-21:
I used the 0.10 version of dependencies on crates.io and applied the example code here. I found that there are significant differences in the APIs, one of which is that the split function in gpio.rs requires different API parameters

Tnze, 2025-08-29:
Why don't they publish new releases to the crate.io😡

liangzhaoyuan12, 2025-08-29:
After I use the dependency for a few days, I found the function: adc_scan_with_dma can’t be used. Sadly, only regular sequences can be used in adc, injection sequences can’t be used. It seems that no one maintains these common functions and API

---- Replied Message ----
| From | ***@***.***> |
| Date | 08/29/2025 15:18 |
| To | stm32-rs/stm32f1xx-hal ***@***.***> |
| Cc | liangzhaoyuan12 ***@***.***>,
Comment ***@***.***> |
| Subject | Re: [stm32-rs/stm32f1xx-hal] Please publish new release to crates.io (Issue #525) |
Tnze left a comment (stm32-rs/stm32f1xx-hal#525)

Why don't they publish new releases to the crate.io😡

burrbull, 2025-08-29:
> What I mainly want is the compatibility with embbeded-hal 1.0, that released jan '24.

This is the main blocker. First of all the SPI implementation:
https://github.com/stm32-rs/stm32f1xx-hal/blob/d6d19c626d0bf7d1be5f76175d7f2facd94cc0c1/src/spi/hal_1.rs#L75

tmehr-enqt, 2025-09-04:
@burrbull what is still missing for embedded-hal 1.0? if there is nothing left I would suggest releasing now and do the other stuff in the next release. Also if you could setup milestones I could see what is missing and help reviewing better.

burrbull, 2025-09-04:
I plan to do release this weekend. I'd also want to take feedback for #422 and #468.

tmehr-enqt, 2025-09-10:
It's not released yet and not on crates.io. @burrbull should I do the release as a draft and you have a look at it? When we decide on a release how do we get it on crates IO? Do we need therealprof or adamgreig for that?

burrbull, 2025-09-10:
https://crates.io/crates/stm32f1xx-hal v0.11.0 updated 1 day ago

PR #548 (2025-09-09): release 0.11
Closes #525
closed 2025-09-10 by tmehr-enqt