stm32-rs/stm32f1xx-hal #532: Mismatched fn names cause the Serial to not work

TinyHai, 2025-05-06:
[stm32f1xx-hal/src/pacext/uart.rs](https://github.com/stm32-rs/stm32f1xx-hal/blob/e9a10d7dd18c18b1ee5903922a83997a76388799/src/pacext/uart.rs#L23)
```Rust
// line 23
wrap_r! {
    pub trait SrR {
        // ...
        // ...
        fn nf(&self) -> usart1::sr::NE_R;
        // ...
        // ...
    }
}
// line 93
mod reg {
    use super::*;

    pub trait SrR: RegisterSpec<Ux = u16> + Readable + Sized {
        // ...
        // ...
        fn nf(r: &R<Self>) -> usart1::sr::NE_R;
        // ...
        // ...
    }
}
// line 151
macro_rules! impl_ext {
    ($(#[$attr:meta])* $uart:ident) => {
        // ...
        impl reg::SrR for $uart::sr::SRrs {
            impl_read! {
                // ...
                nf -> usart1::sr::NE_R;
                // ...
            }
        }
        // ...
    }
}
```
[stm32f1-staging/src/stm32f103/usart1/sr.rs](https://docs.rs/crate/stm32f1-staging/0.20.0/source/src/stm32f103/usart1/sr.rs)
```Rust
impl R {
    // ...
    ///Bit 2 - Noise error flag
    #[inline(always)]
    pub fn ne(&self) -> NE_R {
        NE_R::new(((self.bits >> 2) & 1) != 0)
    }
    // ...
}
```
---
The code that macros are expanded via 'cargo expand'
```Rust
impl<REG: reg::SrR> SrR for R<REG> {
    // ...
    #[inline(always)]
    fn nf(&self) -> usart1::sr::NE_R {
        REG::nf(self)
    }
    // ...
}
impl reg::SrR for usart1::sr::SRrs {
    #[inline(always)]
    fn nf(r: &R<Self>) -> usart1::sr::NE_R {
        r.nf()
    }
}
```
since `nf` can't be found in `R<usart1::sr::SRrs>`, `r.nf()` is actually calling `SrR::nf`, than fall into recursion.

burrbull, 2025-05-06:
> since `nf` can't be found in `R<usart1::sr::SRrs>`, `r.nf()` is actually calling `SrR::nf`, then fall into recursion.

Thank you for founding this. There should be `r.ne()`.
Could you open fix PR.

burrbull, 2025-05-06:
```rust
impl reg::SrR for $uart::sr::SRrs {
            #[inline(always)]
            fn nf(r: &R<Self>) -> usart1::sr::NE_R {
                r.ne()
            }
            impl_read! {
                pe -> usart1::sr::PE_R;
                fe -> usart1::sr::FE_R;
                ore -> usart1::sr::ORE_R;
                idle -> usart1::sr::IDLE_R;
                rxne -> usart1::sr::RXNE_R;
                tc -> usart1::sr::TC_R;
                txe -> usart1::sr::TXE_R;
                lbd -> usart1::sr::LBD_R;
            }
        }
```

TinyHai, 2025-05-06:
> impl reg::SrR for $uart::sr::SRrs {
>             #[inline(always)]
>             fn nf(r: &R<Self>) -> usart1::sr::NE_R {
>                 r.ne()
>             }
>             impl_read! { [...]

I got it, I'll open a PR later

closed 2025-05-07 by TinyHai