stm32-rs/stm32f1xx-hal #539: `dp.BKP.constrain()` not found

iamseol, 2025-08-09:
I am using the version `0.10.0` with `rt`, `stm32f103`, and `medium` features.

The problem I am facing is that cargo keeps saying that `dp.BKP` does not have any methods named `constrain`. But, the file [`blinky_rtc.rs`](https://github.com/stm32-rs/stm32f1xx-hal/blob/master/examples/blinky_rtc.rs) in the examples folder of this repo clearly used RTC in the way I did and I could not figure out how to fix it, so I am creating an issue here. 

In addition, I want to know why the examples... are simply different than what I can run. I mean, it is not just about this, but there were many situations when some people wrote codes using this crate and in the code, they created an instance of some struct with `dp.RCC` as an argument, but every time I tried to test them, Cargo keeps saying that I do not need to put `dp.RCC` as an argument... they did work without it, but I am just wondering why there is such a difference between examples and what I can run.

Here is the code.

```rust
#[cortex_m_rt::entry]
fn main() -> ! {
    let dp = stm32f1xx_hal::pac::Peripherals::take().unwrap();
    let cp = cortex_m::Peripherals::take().unwrap();
    let mut pwr = dp.PWR;
    let mut rcc = dp.RCC.constrain();
    let mut flash = dp.FLASH.constrain();
    let clocks = rcc.cfgr.freeze(&mut flash.acr);

    let mut backup_domain = dp.BKP.constrain(&mut pwr, &mut rcc);
    let mut rtc = Rtc::new(dp.RTC, &mut backup_domain);

    rtc.set_time(0);
    rtc.set_alarm(TOGGLE_INTERVAL_SECONDS);
    rtc.listen_alarm();

    cortex_m::interrupt::free(|cs| *G_RTC.borrow(cs).borrow_mut() = Some(rtc));

    let gpioa = dp.GPIOA.split();

    unsafe {
        cortex_m::peripheral::NVIC::unmask(interrupt::RTCALARM)
    };

    loop {}
}
```

Thank you.

burrbull, 2025-08-09:
There is some unreleased changes.
See examples from corresponding repository state https://github.com/stm32-rs/stm32f1xx-hal/tree/21c64d861c29a97753a22afdd06daaea614592c6 for 0.10.0 release.

iamseol, 2025-08-09:
Oh, gosh. Thank you so much, man.

closed 2025-08-09 by iamseol